ARMmbed/mbed-os #2512: [exporter] Export to KDS results in project file with no references to source

maclobdell, 2016-08-22:
Kinetis Design Studio v3.2
mbed-os-example-blinky at 'mbed-os-5.1.2' release tag.  No local changes.

C:\files\mbed-os-example-blinky> **mbed export -i KDS -m K64F**
Scan: lib
Scan: src
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_CLIENT
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_UVISOR
Scan: FEATURE_IPV4
Scan: FEATURE_IPV6
Scan: FEATURE_STORAGE

Successful:
  * K64F::kds   .\projectfiles\kds_K64F\Unnamed_Project

After importing the project to KDS, and attempting to build, it results in nothing.  Suspect that the paths to the source is invalid.
[image]

[image]

theotherjimmy, 2016-09-09:
If I recall correctly, this could be worked around by moving the project file to the root of the example. The exporters now do that by default. Could you try this again with master, @maclobdell?

maclobdell, 2016-09-12:
This issue is fixed and can be closed.  However, there seems to be other problems building the code in KDS - another issue coming.

closed 2016-09-12 by maclobdell