ARMmbed/mbed-os #2515: Serial putc() in one thread blocked by getc() in another

infinnovation, 2016-08-22:
In an [example program](https://developer.mbed.org/users/infinnovation/code/putc_getc/), one thread sends output on the USB serial console, while another thread loops getting characters from the same Serial instance.  While the getc() is waiting for input, any output from the sender thread is blocked.

To my mind, this seems wrong: the send and receive halves of a serial link should be independent.  The lower level API functions serial_getc and serial_putc do not appear to have this restriction.

sg-, 2016-08-23:
getc() is blocking by default https://developer.mbed.org/forum/mbed/topic/170/
There is a readable() member and some good pointers in that forum post and other linked in it.

How do you suggest the stream implementation for this works?

infinnovation, 2016-08-23:
Thanks for the links.

You're right, I can use readable() (modulo #1987), for example:
```C++
    while (! pc.readable()) {
        Thread::yield();        // Allow other threads to run
    }
    c = pc.getc();
```

Nevertheless, it seems odd that replacing `pc.printf(...)` with `printf(...)` changes behaviour, since the underlying output device is the same.

While it makes sense that one output method should lock out other output methods, and one input method should lock out other input methods, I don't see why input and output could not proceed independently.

closed 2017-10-27 by ChiefBureaucraticOfficer