ARMmbed/mbed-os #2519: NUCLEO_F091RC: tests-mbed_hal-lp_ticker and  tests-mbed_drivers-lp_timeout failing

bulislaw, 2016-08-23:
New test for HAL lp_ticker API and LowPowerTimeout are failing for NUCLEO_F091RC. The IRQ handler for RTC, which backs the implementation on this platform, never fires.

```
mbedgt: checking for GCOV data...
mbedgt: mbed-host-test-runner: stopped and returned 'TIMEOUT'
mbedgt: test on hardware with target id: 075002000753052D38369340
mbedgt: test suite 'tests-mbed_hal-lp_ticker' ........................................................ TIMEOUT in 31.56 sec
	test case: '1ms lp_ticker' ................................................................... SKIPPED in 0.00 sec
	test case: '1s lp_ticker' .................................................................... SKIPPED in 0.00 sec
	test case: '1s lp_ticker deepsleep' .......................................................... SKIPPED in 0.00 sec
	test case: '1s lp_ticker sleep' .............................................................. SKIPPED in 0.00 sec
	test case: '500us lp_ticker' ................................................................. ERROR in 0.00 sec
	test case: '5s lp_ticker' .................................................................... SKIPPED in 0.00 sec
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.7266600495
mbedgt: test suite report:
+-----------------------+---------------+--------------------------+---------+--------------------+-------------+
| target                | platform_name | test suite               | result  | elapsed_time (sec) | copy_method |
+-----------------------+---------------+--------------------------+---------+--------------------+-------------+
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_hal-lp_ticker | TIMEOUT | 31.56              | shell       |
+-----------------------+---------------+--------------------------+---------+--------------------+-------------+
mbedgt: test suite results: 1 TIMEOUT
mbedgt: test case report:
+-----------------------+---------------+--------------------------+------------------------+--------+--------+---------+--------------------+
| target                | platform_name | test suite               | test case              | passed | failed | result  | elapsed_time (sec) |
+-----------------------+---------------+--------------------------+------------------------+--------+--------+---------+--------------------+
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_hal-lp_ticker | 1ms lp_ticker          | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_hal-lp_ticker | 1s lp_ticker           | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_hal-lp_ticker | 1s lp_ticker deepsleep | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_hal-lp_ticker | 1s lp_ticker sleep     | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_hal-lp_ticker | 500us lp_ticker        | 0      | 0      | ERROR   | 0.0                |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_hal-lp_ticker | 5s lp_ticker           | 0      | 0      | SKIPPED | 0.0                |
+-----------------------+---------------+--------------------------+------------------------+--------+--------+---------+--------------------+
mbedgt: test case results: 5 SKIPPED / 1 ERROR
```
```
mbedgt: mbed-host-test-runner: stopped and returned 'TIMEOUT'
mbedgt: test on hardware with target id: 075002000753052D38369340
mbedgt: test suite 'tests-mbed_drivers-lp_timeout' ................................................... TIMEOUT in 31.34 sec
	test case: '1ms LowPowerTimeout' ............................................................. SKIPPED in 0.00 sec
	test case: '1sec LowPowerTimeout' ............................................................ SKIPPED in 0.00 sec
	test case: '1sec LowPowerTimeout from deepsleep' ............................................. SKIPPED in 0.00 sec
	test case: '1sec LowPowerTimeout from sleep' ................................................. SKIPPED in 0.00 sec
	test case: '500us LowPowerTimeout' ........................................................... ERROR in 0.00 sec
	test case: '5sec LowPowerTimeout' ............................................................ SKIPPED in 0.00 sec
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.1424331734
mbedgt: test suite report:
+-----------------------+---------------+-------------------------------+---------+--------------------+-------------+
| target                | platform_name | test suite                    | result  | elapsed_time (sec) | copy_method |
+-----------------------+---------------+-------------------------------+---------+--------------------+-------------+
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_drivers-lp_timeout | TIMEOUT | 31.34              | shell       |
+-----------------------+---------------+-------------------------------+---------+--------------------+-------------+
mbedgt: test suite results: 1 TIMEOUT
mbedgt: test case report:
+-----------------------+---------------+-------------------------------+-------------------------------------+--------+--------+---------+--------------------+
| target                | platform_name | test suite                    | test case                           | passed | failed | result  | elapsed_time (sec) |
+-----------------------+---------------+-------------------------------+-------------------------------------+--------+--------+---------+--------------------+
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_drivers-lp_timeout | 1ms LowPowerTimeout                 | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_drivers-lp_timeout | 1sec LowPowerTimeout                | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_drivers-lp_timeout | 1sec LowPowerTimeout from deepsleep | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_drivers-lp_timeout | 1sec LowPowerTimeout from sleep     | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_drivers-lp_timeout | 500us LowPowerTimeout               | 0      | 0      | ERROR   | 0.0                |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_drivers-lp_timeout | 5sec LowPowerTimeout                | 0      | 0      | SKIPPED | 0.0                |
+-----------------------+---------------+-------------------------------+-------------------------------------+--------+--------+---------+--------------------+
mbedgt: test case results: 5 SKIPPED / 1 ERROR
```

0xc0170, 2016-08-25:
cc @bcostm @adustm @LMESTM

LMESTM, 2016-09-02:
hello - I will give it a try 
cc @jeromecoutant

LMESTM, 2016-09-02:
@svastm

jeromecoutant, 2016-09-02:
Hi

Can you check that the binary file copy has been well performed just before?
Do you see in you logs something like:

	calling mbedhtrun: mbedhtrun -m xxxx
xxx
[COPY][ERR] [ret=1] Command: ['copy', xxx

Thx

sg-, 2017-01-17:
@bulislaw bump

jeromecoutant, 2017-05-22:
Hi

For NUCLEO_F091RC, tests-mbed_drivers-lp_timeout and tests-mbed_hal-lp_ticker tests are OK since mbed-os-5.2.1 with our test bench.

Maybe this issue can be closed ?

Thx

LMESTM, 2017-06-01:
Just tested again with branch being pushed: https://github.com/LMESTM/mbed/tree/STM32_16bits_tickers

+-----------------------+---------------+--------------------------+------------------------+--------+--------+--------+--------------------+
| target                | platform_name | test suite               | test case              | passed | failed | result | elapsed_time (sec) |
+-----------------------+---------------+--------------------------+------------------------+--------+--------+--------+--------------------+
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_hal-lp_ticker | 1ms lp_ticker          | 1      | 0      | OK     | 0.03               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_hal-lp_ticker | 1s lp_ticker           | 1      | 0      | OK     | 1.04               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_hal-lp_ticker | 1s lp_ticker deepsleep | 1      | 0      | OK     | 1.06               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_hal-lp_ticker | 1s lp_ticker sleep     | 1      | 0      | OK     | 1.05               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_hal-lp_ticker | 500us lp_ticker        | 1      | 0      | OK     | 0.05               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_hal-lp_ticker | 5s lp_ticker           | 1      | 0      | OK     | 5.06               |
+-----------------------+---------------+--------------------------+------------------------+--------+--------+--------+--------------------+

+-----------------------+---------------+-------------------------------+-------------------------------------+--------+--------+--------+--------------------+
| target                | platform_name | test suite                    | test case                           | passed | failed | result | elapsed_time (sec) |
+-----------------------+---------------+-------------------------------+-------------------------------------+--------+--------+--------+--------------------+
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_drivers-lp_timeout | 1ms LowPowerTimeout                 | 1      | 0      | OK     | 0.03               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_drivers-lp_timeout | 1sec LowPowerTimeout                | 1      | 0      | OK     | 1.01               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_drivers-lp_timeout | 1sec LowPowerTimeout from deepsleep | 1      | 0      | OK     | 1.08               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_drivers-lp_timeout | 1sec LowPowerTimeout from sleep     | 1      | 0      | OK     | 1.05               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_drivers-lp_timeout | 500us LowPowerTimeout               | 1      | 0      | OK     | 0.05               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | tests-mbed_drivers-lp_timeout | 5sec LowPowerTimeout                | 1      | 0      | OK     | 5.07               |
+-----------------------+---------------+-------------------------------+-------------------------------------+--------+--------+--------+--------------------+

LMESTM, 2017-06-09:
ST_TO_BE_CLOSED
#4424 was merged - @bulislaw  the problem is fixed now, feel free to let us know if any further issue

closed 2017-06-09 by 0xc0170