ARMmbed/mbed-os #2523: TCPServer::accept() hardfaults if maximum number of sockets open

tommikas, 2016-08-23:
1. Open 4 sockets, at least one of which is a TCPServer type socket
2. bind the server socket and set it to listen
3. call `accept()` on the TCPServer socket

sg-, 2016-08-23:
Can you provide a code snippet? This will help increase the time it takes reproduce and fix.

tommikas, 2016-08-24:
I would hope it'd have the opposite effect ;)

Anyway, I hacked mbed-os-example-blinky a bit: https://gist.github.com/tommikas/870cc1a7f50bbf0643d7bc47e6cf5392
(The same result could be achieved by the example in the issue that referenced this one.)

geky, 2016-08-25:
Thanks for reporting this, that is an interesting corner case but should be easy to fail gracefully (returning `NSAPI_ERROR_NO_SOCKETS`)

tommikas, 2016-08-26:
That sounds like a more reasonable way to handle it.

geky, 2016-08-29:
#2579 should resolve this problem as suggested. Let us know if there is still an issue.

tommikas, 2016-09-14:
This has been fixed. Closing.

closed 2016-09-14 by tommikas