ARMmbed/mbed-os #2524: Why GCC code has increased so much?

jeromecoutant, 2016-08-23:
Hi

As example, here is MBED_37 test compiled with GCC_ARM:

MBED release 122: Total Flash memory (text + data + misc): 21053 bytes
MBED release 124: Total Flash memory (text + data + misc): 66559 bytes

Needed flash size has increased by 3!
And this test is now failed for 64KB flash targets...

Regards

Jerome

sg-, 2016-08-23:
Which target in particular? I'm gonna guess it has to do with the change from using newlib-nano to  newlib for thread-safety.

jeromecoutant, 2016-08-24:
I think it is for all target.
Given value is for a STM32F0 target, but here is code size with K64F:
Total Flash memory (text + data + misc): 63434 bytes

closed 2016-08-24 by jeromecoutant

jupe, 2016-08-24:
@jeromecoutant did you accidentally close this issue or what was the conclusion?

jeromecoutant, 2016-08-24:
Hi
This memory size diffference is due to the build option small or standard.

0xc0170, 2016-08-24:
@jeromecoutant Please always provide a reason when closing an issue.