ARMmbed/mbed-os #2537: FEATURE_BLE always gets linked in when compiling for NRF51_DK

tommikas, 2016-08-24:
Using mbed-os-example-blinky as an example, compiling with `mbed compile -t GCC_ARM -m NRF51_DK -c`

```
+----------------------+-------+-------+------+
| Module               | .text | .data | .bss |
+----------------------+-------+-------+------+
| Fill                 |   168 |     7 |   41 |
| Misc                 | 38584 |  2196 | 2149 |
| features/FEATURE_BLE | 15483 |     9 |  532 |
| features/frameworks  |  3845 |    92 | 1008 |
| hal/common           |  3165 |     4 |  333 |
| hal/targets          | 14489 |     0 | 1423 |
| rtos/rtos            |   131 |     4 |    0 |
| rtos/rtx             |  5671 |    20 | 4226 |
| Subtotals            | 81536 |  2332 | 9712 |
+----------------------+-------+-------+------+
Allocated Heap: 6632 bytes
Allocated Stack: 2048 bytes
Total Static RAM memory (data + bss): 12044 bytes
Total RAM memory (data + bss + heap + stack): 20724 bytes
Total Flash memory (text + data + misc): 83868 bytes
Image: ./.build/NRF51_DK/GCC_ARM/mbed-os-example-blinky.hex
```

Then add `mbed-os/features/FEATURE_BLE` into .mbedignore and compile again:

```
+---------------------+-------+-------+------+
| Module              | .text | .data | .bss |
+---------------------+-------+-------+------+
| Fill                |    76 |     4 |   21 |
| Misc                | 29715 |  2188 |   85 |
| features/frameworks |  3853 |    92 | 1008 |
| hal/common          |  3123 |     4 |  333 |
| hal/targets         |  7991 |     0 |  871 |
| rtos/rtos           |   131 |     4 |    0 |
| rtos/rtx            |  5671 |    20 | 4226 |
| Subtotals           | 50560 |  2312 | 6544 |
+---------------------+-------+-------+------+
Allocated Heap: 9824 bytes
Allocated Stack: 2048 bytes
Total Static RAM memory (data + bss): 8856 bytes
Total RAM memory (data + bss + heap + stack): 20728 bytes
Total Flash memory (text + data + misc): 52872 bytes
Image: ./.build/NRF51_DK/GCC_ARM/mbed-os-example-blinky.hex
```

The application doesn't use any of the feature, but it's getting linked in anyway. Should this be happening?

tommikas, 2016-08-24:
http://jira.arm.com/browse/IOTSFW-2870

0xc0170, 2016-08-25:
cc @pan-

PR #2778 (2016-09-26): BLE - Nordic implementation: Move singleton into a function.
This change remove BLE code from the binary generated if the user code doesn't
use BLE.

It fixes #2537.

Note: it was chosen to not use the SingletonPtr class in order to keep the code compatible with mbed OS 3. The BLE API is not thread safe, this is not an issue in this code.
closed 2016-09-26 by sg-