ARMmbed/mbed-os #2553: TCPSocket stream stops sending/receiving after a while

infinnovation, 2016-08-26:
A [test program](https://developer.mbed.org/users/infinnovation/code/socket-test/) implements a simple TCP echo server: after accepting a connection, it repeatedly reads a chunk of data then sends it back on the same socket.  For each chunk received or sent, it outputs a character to the serial console.  Additionally, a thread prints a message every second (so I can verify that the board is still alive).

To test, I connect and send a stream of zero bytes:
```
  nc 10.212.1.150 7 < /dev/zero > /dev/null
```
What I expect is that the stream continues indefinitely (albeit slowly, given the small TCP window size).

What I see is that the stream continues for some seconds, usually with some short stalls (which is allowable), but then stops completely.  In this state, the board does not respond to ping.

The problem appears to be affected by other threads: if I disable the tick thread, the echo stream may continue for a long time (minutes?) before failing.  The last successful action was a recv() rather than a send().

If I enable the console statistics-viewing thread, I see that the link is still running - it shows increasing numbers of dropped packets but no more received.

geky, 2016-08-26:
Thanks for reporting the issue and the good test case. This will need some looking into.

infinnovation, 2016-09-13:
The stall may be triggered by other traffic on the network e.g. ARP requests.  That might explain the somewhat random nature of the problem.  Needs some more investigation with wireshark.

sg-, 2016-09-19:
> Needs some more investigation with wireshark. 

@infinnovation any update?

infinnovation, 2016-09-20:
Apologies for the somewhat vague hand-waving while trying to pin down a reproducible fault condition.
I haven't had an opportunity to progress this further due to other demands on my time.  Maybe in the next week or two...

sg-, 2017-01-17:
@infinnovation any update?

infinnovation, 2017-01-19:
@sg- I don't anticipate having the opportunity to look at this for several weeks yet.

closed 2017-10-27 by ChiefBureaucraticOfficer