ARMmbed/mbed-os #2554: TCPSocket connection close by remote end reported as device error

infinnovation, 2016-08-26:
Using a [test program](https://developer.mbed.org/users/infinnovation/code/socket-test/) (an echo server), I connect from a Linux PC and send a stream of data:
```
nc $K64F_IP_ADDRESS 7 < /dev/zero > /dev/null
```
and then kill the nc process with a Ctrl-C

The board gets an error code -3011 NSAPI_ERROR_DEVICE_ERROR "failure interfacing with the network procesor" (sic).  This can come either from the recv() call or from the send() call - I guess depending on the timing of the event relative to the send/recv loop.

I would expect either a 0 return from recv() to indicate end of stream; or some other error from send(), maybe NSAPI_ERROR_NO_CONNECTION.

infinnovation, 2016-08-26:
The underlying error in the lwip library is ERR_RST

geky, 2016-08-26:
Thanks for reporting this, looks like it was just a short-sight in handing lwip's error codes:
https://github.com/ARMmbed/mbed-os/blob/master/features/net/FEATURE_IPV4/lwip-interface/lwip_stack.c#L213

closed 2016-09-18 by sg-