ARMmbed/mbed-os #2560: LWIP disables MPU during initialization and breaks uVisor

sherrellbc, 2016-08-26:
`MPU->CESR &= ~MPU_CESR_VLD_MASK;`

https://github.com/ARMmbed/mbed-os/blob/master/features/net/FEATURE_IPV4/lwip-interface/lwip-eth/arch/TARGET_Freescale/hardware_init_MK64F12.c#L41

The MPU appears to be a protected memory region that cannot be accessed by unprivileged code. Access to this region results in a halt with uVisor. 

My original post was over at [uVisor](https://github.com/ARMmbed/uvisor) as I originally thought it was a configuration related issue. See t here: https://github.com/ARMmbed/uvisor/issues/315

geky, 2016-08-29:
Thanks for reporting this issue. I went ahead and continued the discussion on https://github.com/ARMmbed/uvisor/issues/315 to keep the progress on the issue consolidated.

sg-, 2016-09-19:
This should be fixed with https://github.com/ARMmbed/mbed-os/pull/2733

sherrellbc, 2016-09-26:
@sg- 

Although https://github.com/ARMmbed/mbed-os/pull/2733 introduced support for LwIP using DMA, it alone was not a sufficient patch to allow LwIP to fully function. Take a look at [my original](https://github.com/ARMmbed/uvisor/issues/315) issue; there were some run-time problems with memory management of mutexes, or similar.

The actual fix came with https://github.com/ARMmbed/mbed-os/commit/2a42255bba42271366703ce083bcab37374bcce6, but so many changes were introduced that it is hard to say which patch corrected the offending code.

closed 2016-09-26 by sherrellbc