ARMmbed/mbed-os #2565: incompatible redefinition of macro "UNITY_INCLUDE_CONFIG_H"

maclobdell, 2016-08-27:
from a fresh import of this example 
release tag mbed-os-5.1.2

there are hundreds of warnings for:

This issue was found while testing ARMmbed/mbed-os-example-client
copy of: https://github.com/ARMmbed/mbed-os-example-client/issues/77

[Warning] mbed_config.h@29,0: #47-D: incompatible redefinition of macro "UNITY_INCLUDE_CONFIG_H" (declared at line 28)

in mbed_config.h, multiple defines are generated for this macro.

// Macros
#define UNITY_INCLUDE_CONFIG_H None // defined by library:utest
#define UNITY_INCLUDE_CONFIG_H // defined by library:utest

tommikas, 2016-08-29:
I think this might be the same problem as in #2486, which was fixed in master in #2492 and #2494.

maclobdell, 2016-09-22:
this is resolved

closed 2016-09-22 by maclobdell