ARMmbed/mbed-os #2567: mbed TLS example does not run on K64F, RTX error code

janjongboom, 2016-08-29:
We have the following program http://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-tls-tls-client/ that uses mbed TLS to connect to an HTTPS server.

On Twitter we received the following message: [image].

```
RTX error code: 0x00000001, task ID: 0x2000F184
```

When I run this example on my own K64F (against mbed OS 5.1 and mbed OS master) I get:

```
mbed assertation failed: interrupts_disabled, file: ./mbed-os/hal/common/mbed_critical.c, line 54 
```

@pjbakker does not have a clue either...

c1728p9, 2016-08-29:
Hi @janjongboom ``RTX error code: 0x00000001, task ID: 0x2000F184`` indicates a stack overflow occurred. The assert indicates that something turned on interrupts while in a critical section. I'll try to reproduce this today if I get time.

c1728p9, 2016-08-29:
I can reproduce this locally, and it is for sure a stack overflow. It also looks like the crtitical section counter is getting clobbered by this stack overflow, which occasionally leads to the failed assert you see. I opened an issue to track this.

closed 2016-09-19 by sg-