ARMmbed/mbed-os #2569: FEATURE_CLIENT compilation fails

tommikas, 2016-08-29:
```
Compile: m2mconnectionhandler.cpp
Compile: m2mtimer.cpp
Compile: m2mtimerpimpl.cpp
Compile: m2mconnectionhandlerpimpl.cpp
[Error] m2mconnectionhandlerpimpl.cpp@41,8: 'MemoryPool' does not name a type
[Error] m2mconnectionhandlerpimpl.cpp@84,9: 'memory_pool' was not declared in this scope
[Error] m2mconnectionhandlerpimpl.cpp@159,28: 'memory_pool' was not declared in this scope
[Error] m2mconnectionhandlerpimpl.cpp@266,28: 'memory_pool' was not declared in this scope
[Error] m2mconnectionhandlerpimpl.cpp@328,28: 'memory_pool' was not declared in this scope
[ERROR] ./mbed-os/features/FEATURE_CLIENT/mbed-client-classic/source/m2mconnectionhandlerpimpl.cpp:41:8: error: 'MemoryPool' does not name a type
 static MemoryPool<M2MConnectionHandlerPimpl::TaskIdentifier, MBED_CLIENT_EVENT_LOOP_SIZE/64> memory_pool;
        ^
./mbed-os/features/FEATURE_CLIENT/mbed-client-classic/source/m2mconnectionhandlerpimpl.cpp: In function 'void connection_tasklet_event_handler(arm_event_s*)':
./mbed-os/features/FEATURE_CLIENT/mbed-client-classic/source/m2mconnectionhandlerpimpl.cpp:84:9: error: 'memory_pool' was not declared in this scope
         memory_pool.free(task_id);
         ^
```

Since https://github.com/ARMmbed/mbed-os/pull/2527 was merged in, because of https://github.com/ARMmbed/mbed-os/pull/2527/files#diff-d62a3ea84a2a42fc1bf16f8fa83b4725L30

yogpan01, 2016-08-29:
This is because of 2 merges which missed the include header file.
There needs new PR from mbed-client-classic to resolve this issue. I will create a new PR now to get this resolved.

tommikas, 2016-08-30:
Fixed in #2570, so closing this.

closed 2016-08-30 by tommikas