ARMmbed/mbed-os #2574: Scope of HAL layer

Mohamedsaleh14, 2016-08-29:
There's no document describes the scope of HAL layer, I need to find documentation where I can find the range of input parameters,  the expected behavior of the function, what's the Common layer expecting HAL layer to do exactly and it's scope of functionality?

In other word, I need HLD(High level design), mHLD (module high level design) mapped to SRS that describe the static and dynamic behavior of the mbed and its modules, software requirements, Software component design

Those documentation will be beneficial for porting mbed to any target, and not limit mbed to the current hardware boards recommended on mbed website

sherrellbc, 2016-08-29:
This guide may help.

https://docs.mbed.com/docs/uvisor-and-uvisor-lib-documentation/en/latest/Porting/

Mohamedsaleh14, 2016-08-29:
Thank you sherrellbc ,, I am searching for complete design documentation, If you followed the implementation of HAL layer of many targets provided by mbed, you'll find major differences in implementation, which emphasize different behavior of mbed with different targets, And that's not what I expect. There should be software requirements I can use, to be able to design my application according to, and accurately determine its behavior

0xc0170, 2016-08-31:
+1 for having HLD and LLD, they would be beneficial.

We would like to add tests for HAL, the LLD should come with it (might be earlier). We keep this updated with a progress

closed 2017-10-27 by ChiefBureaucraticOfficer