ARMmbed/mbed-os #2591: All exporters broken with mbed OS 5

janjongboom, 2016-08-31:
mbed OS 5, online IDE, K64F target:

- Export to Keil uVision 4 -> creates a build which does not include mbed-os repo
- Export to GCC code sourcery -> does not download anything
- Export to GCC ARM embedded -> does not download anything
- IAR systems -> creates a build without mbed-os repo
- KDS -> creates a build without mbed-os repo
- uVision 4 (why the f* do we have uVision 4 twice?!) -> creates a build w/o mbed-os
- uVision 5 -> Same story
- Emblocks -> See above
- ZIP file -> also no mbed-os

This looks very bad. At least disable them if we know they are broken.

0xc0170, 2016-08-31:
cc @screamerbg 

Note: This issue tracker is not for the online IDE

0xc0170, 2016-08-31:
@janjongboom if you test offline exporting, mbed-os-5.1.2 for instance, does that work? just to confirm it's online IDE issue (not tools that are part of this repository)

janjongboom, 2016-08-31:
@0xc0170 No, same issues.

janjongboom, 2016-08-31:
According to @screamerbg we have reworked exporters currently in beta. So hopefully this will be solved in a week or so :-)

ytsuboi, 2016-09-09:
How is reworking exporters going? This is huge problem. :-(

janjongboom, 2016-09-09:
@ytsuboi Here's the PR: #2526.

janjongboom, 2016-09-09:
@ytsuboi Here's the PR: #2245.

theotherjimmy, 2016-09-15:
Hey, so that PR is merged and the exporters are live on the website. The should even be backward compatible with old mbed revisions. The zip exporter may time out on you as it may have to create a 600 MB zip, which takes time.

janjongboom, 2016-09-15:
@theotherjimmy Thanks. I'll put something on the blog tomorrow morning.

theotherjimmy, 2016-09-15:
for good measure, you should test them before you announce them fixed.

sg-, 2016-09-19:
@janjongboom Can this be closed?