ARMmbed/mbed-os #2598: Multiple tickers stop working after several seconds in a high load

aetheln, 2016-09-01:
Working on a STM32F030 system. The issue is:  start some tickers with short period ( a few milliseconds) do things like control LED arrays via SPI/I2C, read/write from serial etc. After a while, some, maybe all tickers just stop calling the handlers. Attach debugger to check the system, the system still working, no hard-fault things etc.
I checked the code and found that on some st platforms using 16-bit timer there is a check for the past event in function us_ticker_set_interrupt
```
    if (delta <= 0) { // This event was in the past
        us_ticker_irq_handler();
    } else {
      ...
```
when the condition happens, it will cause processing of the event immediately. In the Ticker class, the processor function will schedule the next event, the calling path may trigger a recursive call along the calling path of ticker_insert_event, us_ticker_set_interrupt, us_ticker_irq_handler ..., In a high load system, some ticker event may miss it's window and cause the condition to happen. But in the ticker_insert_event function, the following piece of code have one reentry issue:
```
    /* if prev is NULL we're at the head */
    if (prev == NULL) {
        data->queue->head = obj;
        data->interface->set_interrupt(timestamp);
    } else {
        prev->next = obj;
    }
    /* if we're at the end p will be NULL, which is correct */
    obj->next = p;
```
the obj->next pointer modification is after set_interrupt. When the reentry condition  happens, the pointer final value will be wrong.
A simple fix here is to move `obj->next = p;` before set_interrupt.
Also it might be better for the st hal us_ticker_set_interrupt function to avoid call to the irq handler by scheduling the handling in a very near future, for example, 1 us later.

aetheln, 2016-09-01:
as requested by @0xc0170, I open this new ticket to track this issue (originally as a comment to another issue)

0xc0170, 2016-09-01:
cc @bcostm @adustm @LMESTM

bittailor, 2017-01-03:
All the _TARGET_SAM_CortexM0P_ targets have the same issue, since they also call the IRQ handler directly inside us_ticker_set_interrupt when the event already occurred in the past:

> mbed-os/targets/TARGET_Atmel/TARGET_SAM_CortexM0P/us_ticker.c
```cpp
void us_ticker_set_interrupt(timestamp_t timestamp)
{
    uint32_t cur_time;
    int32_t delta;

    cur_time = us_ticker_read();
    delta = (int32_t)((uint32_t)timestamp - cur_time);
    if (delta < 0) {
        /* Event already occurred in past */
        us_ticker_irq_handler();
        return;
    }

    NVIC_DisableIRQ(TICKER_COUNTER_IRQn);
    NVIC_SetVector(TICKER_COUNTER_IRQn, (uint32_t)TICKER_COUNTER_Handlr);

    /* Enable the callback */
    tc_enable_callback(&us_ticker_module, TC_CALLBACK_CC_CHANNEL0);
    tc_set_compare_value(&us_ticker_module, TC_COMPARE_CAPTURE_CHANNEL_0, (uint32_t)timestamp);

    NVIC_EnableIRQ(TICKER_COUNTER_IRQn);
}
```

I run into this problem by using mbed-events on a _SAMD21G18A_ target:

```cpp
#include "mbed.h"
#include "mbed_events.h"

DigitalOut led(LED1,0);
EventQueue eventQueue(32 * EVENTS_EVENT_SIZE);

void subtask(int id) {
   printf("subtask %d\n", id);
}

void task() {
   static int counter = 0;
   led = !led;
   printf("[%lu] task %d\n", us_ticker_read()/1000, counter++);
   const int numberOfSubtasks = 6;
   for(int i = 0 ; i < numberOfSubtasks ; i++){
      eventQueue.call_in(5, &subtask, i);
   }
}

int main() {
   printf("</main>\n");
   eventQueue.call_every(200, &task);
   eventQueue.dispatch_forever();
   printf("</main>\n");
}
```

This stops working after about 1/2 hour. The problem is that mbed-events uses a Ticker to avoid Timer overflows but that ticker does not work because of the Ticker issue and then the Timer overflows and the periodic event stops working.

I currently did this workaround:

```diff
diff --git a/targets/TARGET_Atmel/TARGET_SAM_CortexM0P/us_ticker.c b/targets/TARGET_Atmel/TARGET_SAM_CortexM0P/us_ticker.c
index 4fb0880..da8ba67 100644
--- a/targets/TARGET_Atmel/TARGET_SAM_CortexM0P/us_ticker.c
+++ b/targets/TARGET_Atmel/TARGET_SAM_CortexM0P/us_ticker.c
@@ -133,9 +133,8 @@ void us_ticker_set_interrupt(timestamp_t timestamp)
     cur_time = us_ticker_read();
     delta = (int32_t)((uint32_t)timestamp - cur_time);
     if (delta < 0) {
-        /* Event already occurred in past */
-        us_ticker_irq_handler();
-        return;
+       /* Event already occurred in past */
+       timestamp = us_ticker_read() + 100;
     }

     NVIC_DisableIRQ(TICKER_COUNTER_IRQn);
```

which works on my board, but feels dirty, since I just use an arbitrary _delay_ that works for my board. 

I did not manage to trigger the `us_ticker_irq_handler` immediately by setting the interrupt as pending (`NVIC_SetPendingIRQ`) since the TC handler (`_tc_interrupt_handler`) checks some TC related registers to determine which callbacks should be executed and they do not match when when I just set the interrupt pending flag. 

So I would  appreciate any help to solve this in a clean way for the _TARGET_SAM_CortexM0P_ targets.

0xc0170, 2017-01-04:
@bittailor Thanks for the detailed description. As I undestand you were not able to make it work similar to other implementations, for instance for K64F:

```
    if (delta <= 0) {
        // This event was in the past.
        // Set the interrupt as pending, but don't process it here.
        // This prevents a recurive loop under heavy load
        // which can lead to a stack overflow.
        NVIC_SetPendingIRQ(PIT3_IRQn);
        return;
    }
```

cc @kpurusho @akhilpanayamparambil @vimalrajr (for the above issue)

bittailor, 2017-01-04:
Yes, I tried 
```cpp
void us_ticker_set_interrupt(timestamp_t timestamp)
{
    uint32_t cur_time;
    int32_t delta;

    cur_time = us_ticker_read();
    delta = (int32_t)((uint32_t)timestamp - cur_time);
    if (delta < 0) {
        /* Event already occurred in past */
        NVIC_SetPendingIRQ(TICKER_COUNTER_IRQn);
        return;
    }

    NVIC_DisableIRQ(TICKER_COUNTER_IRQn);
    NVIC_SetVector(TICKER_COUNTER_IRQn, (uint32_t)TICKER_COUNTER_Handlr);

    /* Enable the callback */
    tc_enable_callback(&us_ticker_module, TC_CALLBACK_CC_CHANNEL0);
    tc_set_compare_value(&us_ticker_module, TC_COMPARE_CAPTURE_CHANNEL_0, (uint32_t)timestamp);

    NVIC_EnableIRQ(TICKER_COUNTER_IRQn);
}
```

put this did not work. 

The interrupt is triggered and the `TC4_Handler` is called and calls `_tc_interrupt_handler` which checks the TC register to call the appropriate callback(s) but since the TC register is not set (interrupt was triggered by software and not the TC hardware) none of the callbacks is triggered. 

> mbed-os/targets/TARGET_Atmel/TARGET_SAM_CortexM0P/drivers/tc/tc_interrupt.c
```cpp
void _tc_interrupt_handler(
    uint8_t instance)
{
    /* Temporary variable */
    uint8_t interrupt_and_callback_status_mask;

    /* Get device instance from the look-up table */
    struct tc_module *module
        = (struct tc_module *)_tc_instances[instance];

    /* Read and mask interrupt flag register */
    interrupt_and_callback_status_mask = module->hw->COUNT8.INTFLAG.reg &
                                         module->register_callback_mask &
                                         module->enable_callback_mask;

    /* Check if an Overflow interrupt has occurred */
    if (interrupt_and_callback_status_mask & TC_INTFLAG_OVF) {
        /* Invoke registered and enabled callback function */
        (module->callback[TC_CALLBACK_OVERFLOW])(module);
        /* Clear interrupt flag */
        module->hw->COUNT8.INTFLAG.reg = TC_INTFLAG_OVF;
    }

    /* Check if an Error interrupt has occurred */
    if (interrupt_and_callback_status_mask & TC_INTFLAG_ERR) {
        /* Invoke registered and enabled callback function */
        (module->callback[TC_CALLBACK_ERROR])(module);
        /* Clear interrupt flag */
        module->hw->COUNT8.INTFLAG.reg = TC_INTFLAG_ERR;
    }

    /* Check if an Match/Capture Channel 0 interrupt has occurred */
    if (interrupt_and_callback_status_mask & TC_INTFLAG_MC(1)) {
        /* Invoke registered and enabled callback function */
        (module->callback[TC_CALLBACK_CC_CHANNEL0])(module);
        /* Clear interrupt flag */
        module->hw->COUNT8.INTFLAG.reg = TC_INTFLAG_MC(1);
    }

    /* Check if an Match/Capture Channel 1 interrupt has occurred */
    if (interrupt_and_callback_status_mask & TC_INTFLAG_MC(2)) {
        /* Invoke registered and enabled callback function */
        (module->callback[TC_CALLBACK_CC_CHANNEL1])(module);
        /* Clear interrupt flag */
        module->hw->COUNT8.INTFLAG.reg = TC_INTFLAG_MC(2);
    }
}
```

I also tried to change the interrupt handler by providing my own handler function in this special case:

```cpp

void us_ticker_my_irq_handler_internal(void)
{
    us_ticker_irq_handler();
}

void us_ticker_set_interrupt(timestamp_t timestamp)
{
    uint32_t cur_time;
    int32_t delta;

    cur_time = us_ticker_read();
    delta = (int32_t)((uint32_t)timestamp - cur_time);
    if (delta < 0) {
        /* Event already occurred in past */
        NVIC_SetVector(TICKER_COUNTER_IRQn, (uint32_t)us_ticker_my_irq_handler_internal);
        NVIC_SetPendingIRQ(TICKER_COUNTER_IRQn);
        return;
    }

    NVIC_DisableIRQ(TICKER_COUNTER_IRQn);
    NVIC_SetVector(TICKER_COUNTER_IRQn, (uint32_t)TICKER_COUNTER_Handlr);

    /* Enable the callback */
    tc_enable_callback(&us_ticker_module, TC_CALLBACK_CC_CHANNEL0);
    tc_set_compare_value(&us_ticker_module, TC_COMPARE_CAPTURE_CHANNEL_0, (uint32_t)timestamp);

    NVIC_EnableIRQ(TICKER_COUNTER_IRQn);
}
```
But this also did not work. 
Even with `NVIC_SetVector(TICKER_COUNTER_IRQn, (uint32_t)us_ticker_my_irq_handler_internal);`  still the original handler `TC4_Handler` is called by the interrupt. I do not know why, maybe I misunderstood `NVIC_SetVector()`

bittailor, 2017-01-04:
This [Gist](https://gist.github.com/bittailor/5ac36c2f33c5d0495cc68f54155b71df) quickly reproduces the problem on a _SAMD21G18A_ target.

I also had to improve my dirty workaround since us_ticker_read() changes from when its compared with the timestamp until the timestamp is set as TC compare value.

```diff
 void us_ticker_set_interrupt(timestamp_t timestamp)
 {
+    #define MIN_DELTA 100
+
     uint32_t cur_time;
     int32_t delta;

     cur_time = us_ticker_read();
     delta = (int32_t)((uint32_t)timestamp - cur_time);
-    if (delta < 0) {
-        /* Event already occurred in past */
-        us_ticker_irq_handler();
-        return;
+    if (delta < MIN_DELTA) {
+       /* Event already occurred in past */
+       timestamp = us_ticker_read() + MIN_DELTA;
     }

     NVIC_DisableIRQ(TICKER_COUNTER_IRQn);
```

akhilpanayamparambil, 2017-01-09:
@bittailor How did it behave after the work around mentioned above ?

bittailor, 2017-01-09:
@akhilpanayamparambil For my application and board the workaround fixed the problem and the code run for more than a day without any issue. 

What I don't like in my workaround is the timing I need to use (`#define MIN_DELTA 100`), since this probably depends on CPU frequency and other things. I tried different values with my board and with a value of 50 it started to worked so I duplicated the value to have some reserve.

akhilpanayamparambil, 2017-01-09:
@bittailor We will have to find out the time taken for one CPU operation with the current frequency and arrive at a optimum value (Approx CPU cycles in between these instructions * Time taken for one CPU operation) in the run time instead of using MIN_DELTA. This will make the implementation more complex and the time taken shall be added up once we add this piece of code. So its better to use your implementation and go forward with a optimum value for MIN_DELTA estimating the number of instructions involved in between.

akhilpanayamparambil, 2017-01-09:
cc @Parthasarathy

LMESTM, 2017-06-01:
@aetheln  @0xc0170 
I've pushed a branch with fixes that will apply to F0 family so should fix this issue.
The point is to trigger the event so that it will eb handled in the interrupt, and not call directly the interrupt handler. 
Here is the branch with fixes to test: https://github.com/LMESTM/mbed/tree/STM32_16bits_tickers
Anyone could help and confirm the fix is ok ? there is no explanation here on how to reproduce the issue

LMESTM, 2017-06-09:
ST_TO_BE_CLOSED
#4424 was merged - this problem should fixed now - past events handling is delayed to not be called directly from interrupt context 
feel free to let us know if any further issue

closed 2017-06-09 by 0xc0170