ARMmbed/mbed-os #2605: "Verbose" mode for memap.py

javier-moreno-tridonic-com, 2016-09-02:
Note: This is just a template, so feel free to use/remove the unnecessary things

## Description

Type: Feature request
Priority: Minor
Feature Request
---

To analyse memory usage and identify possible optimisation points, I've been modifying memap.py in the following manner:

original code
---
```python
    @staticmethod
    def path_object_to_module_name(txt):
        """ Parse a path to object file to extract it's module and object data
        Positional arguments:
        txt - the path to parse the object and module name from
        """

        txt = txt.replace('\\', '/')
        rex_mbed_os_name = r'^.+mbed-os\/(.+)\/(.+\.o)$'
        test_rex_mbed_os_name = re.match(rex_mbed_os_name, txt)

        if test_rex_mbed_os_name:

            object_name = test_rex_mbed_os_name.group(2)
            data = test_rex_mbed_os_name.group(1).split('/')
            ndata = len(data)

            if ndata == 1:
                module_name = data[0]
            else:
                module_name = data[0] + '/' + data[1]

            return [module_name, object_name]
        else:
            return ['Misc', ""]
```

My code
---
```python
    @staticmethod
    def path_object_to_module_name(txt):
        """ Parse a path to object file to extract it's module and object data

        Positional arguments:
        txt - the path to parse the object and module name from
        """

        txt = txt.replace('\\', '/')
        
        # print('txt: ' + txt)
        
        rex_mbed_os_name = r'^.+mbed-os\/(.+)\/(.+\.o)$'
        test_rex_mbed_os_name = re.match(rex_mbed_os_name, txt)

        if test_rex_mbed_os_name:
            object_name = test_rex_mbed_os_name.group(2)
            data = test_rex_mbed_os_name.group(1).split('/')
            ndata = len(data)

            if ndata == 1:
                module_name = data[0]
            else:
                module_name = data[0] + '/' + data[1]

            return [module_name, object_name]
[... 189 lines cut ...]
| libstdc++.a(guard_error.o)        |    115 |     0 |     0 |
| libstdc++.a(locale-inst.o)        |    356 |     4 |    48 |
| libstdc++.a(locale_facets.o)      |    136 |     4 |     0 |
| libstdc++.a(new_handler.o)        |     16 |     0 |     4 |
| libstdc++.a(new_op.o)             |     68 |     0 |     0 |
| libstdc++.a(new_opv.o)            |     20 |     0 |     0 |
| libstdc++.a(si_class_type_info.o) |    381 |     0 |     0 |
| libstdc++.a(snprintf_lite.o)      |    396 |     0 |     0 |
| libstdc++.a(stdexcept.o)          |    402 |     0 |     0 |
| libstdc++.a(string-inst.o)        |   1337 |     0 |    16 |
| libstdc++.a(system_error.o)       |    473 |    12 |     0 |
| libstdc++.a(tinfo.o)              |     77 |     0 |     0 |
| libstdc++.a(wlocale-inst.o)       |    196 |     4 |    48 |
| main.o                            |   4285 |    21 |   410 |
| rtos/rtos                         |    575 |     4 |     0 |
| rtos/rtx                          |   7037 |    20 |  2683 |
| Subtotals                         | 386365 |  3008 | 39488 |

Allocated Heap: unknown
Allocated Stack: unknown
Total Static RAM memory (data + bss): 42496 bytes
Total RAM memory (data + bss + heap + stack): 42496 bytes
Total Flash memory (text + data + misc): 389373 bytes

I think it would be nice to have such print as optional for the mbed-cli (maybe even for mbed-os features so we could help you identify and fix issues).

MarceloSalazar, 2016-09-02:
Hi Javier,

This feature looks very useful :)
A few suggestions:

- Add a verbose option to generate more information when invoking memap from the command line (by default should be disabled): https://github.com/ARMmbed/mbed-os/blob/5c82db820357508a27f7c6a18d90dc1d081851af/tools/memap.py#L597

- Display `Misc` to make the output more consistent with the default behavior:
 `libc.a(lib_a-__atexit.o)` --> `Misc\libc.a(lib_a-__atexit.o)`

It would be great if you could send us a PR and we'll be happy to review it.

Thanks,
Marcelo

javier-moreno-tridonic-com, 2016-09-05:
Well I'd be happy to prepare it, just I would need some pointers for this:

- Should be this a memap only feature? I mean, should this feature be also accessible by using a "mbed compile" flag? If so, any flag name suggestion that can be shared across both scripts?

MarceloSalazar, 2016-09-05:
I'd suggest this to be a memap only feature for now (e.g. invoke the tool using `python memap ...`).
This will let us prepare the interfaces and do further testing. We will consider calling it from the mbed compile script.

nuket, 2016-09-07:
This is great, I'm also looking for a way to have a deeper view into flash / stack / heap memory allocations.

meriac, 2016-09-20:
@nuket : I had good results parsing arm-non-eabi-nm output - feel free to have a look on an early prototype for [d3.js based linker visualization](https://github.com/ARMmbed/mbed-os-linker-report).

You can find the command line settings I chose [here](https://github.com/ARMmbed/mbed-os-linker-report/blob/master/elfsize.php#L20-L22). Please note that I had to enable debug information to get line number resolution in the mbed release build.
You can find a [quick patch for GCC](https://github.com/ARMmbed/mbed-os-linker-report/blob/master/patches/0001-GCC-Always-enable-debug-information.patch) in my tool repo. The official pull request is [#2139](https://github.com/ARMmbed/mbed-os/pull/2139) for adding debug symbol support to all mbed supported release builds.

javier-moreno-tridonic-com, 2016-10-24:
Since this functionality was merged already (see https://github.com/ARMmbed/mbed-os/pull/2845) this can be closed

closed 2016-10-24 by javier-moreno-tridonic-com