ARMmbed/mbed-os #2606: Better handling for out of event pool

ccli8, 2016-09-02:
I am testing the sample `mbed-os-example-client` with WiFi module `esp8266` on target board `NUMAKER_PFM_NUC472`.
**Target baord: NUMAKER_PFM_NUC472**
**mbed-os-example-client (SHA code: d22a0db8803811eec7817c770f52c3386e3cc7ab)**
**mbed-os (SHA code: 9111aa4c2dc74fd9bc29f6c37a26489c09cf8a68)**
**esp8266-driver (SHA code: 29d63ae2ee0a233e2fbd9577cdddc7661bb783d1)**

After fixing some target related issues, it can work normally. For my development and debug requirement, I add printf calls in crypto related functions in `mbed-os` library. This will cause system crash sometimes and no message tells the cause. I try to export it to KEIL uvision project and check through debugger. Per debugger call stack, the cause is found out. The reduced call stack (from bottom to top) would be: 
1. `ESP8266Interface::event` called from serial interrupt handler
2. `M2MConnectionHandlerPimpl::socket_event` called due to socket state change
3. `M2MInterfaceImpl::socket_error` called as `memory_pool.alloc` fails in `M2MConnectionHandlerPimpl::socket_event`. If `memory_pool.alloc` succeeds, no error.
4. `_mutex_acquire` finally called and it causes Memory Management Fault because now it is in interrupt context
```
void M2MConnectionHandlerPimpl::socket_event()
{
    TaskIdentifier* task = memory_pool.alloc();
    if (!task) {
    	_observer.socket_error(M2MConnectionHandler::SOCKET_READ_ERROR,** true);
        return;
    }
    task->pimpl = this;

   ......
}
```
I suggest better handling as event pool runs out such as below. Note `M2MConnectionHandlerPimpl::socket_event` is called in interrupt context.
1. No such Memory Management Fault
2. Message could be output to tell the cause

yogpan01, 2016-09-02:
Thanks for reporting. We are fixing this at the moment.

sg-, 2017-01-16:
@yogpan01 bump

sg-, 2017-01-16:
No longer in mbed OS. @ccli8 please file this issue against mbed-os-example-client if its not resolved.

closed 2017-01-16 by sg-