ARMmbed/mbed-os #2608: '--build' option breaks memory table generation for at least ARM and IAR toolchains

tommikas, 2016-09-02:
For example when compiling mbed-os-example-blinky

`mbed compile --build .build/K64F_ARM_BLINKY/ -t ARM -m K64F -j0 -c`

```
Link: mbed-os-example-blinky
Elf2Bin: mbed-os-example-blinky
Warning: specified toolchain doesn't match with path to the memory map file.
+-----------+-------+-------+-------+
| Module    | .text | .data |  .bss |
+-----------+-------+-------+-------+
| Misc      | 41946 |   792 | 10288 |
| Subtotals | 41946 |   792 | 10288 |
+-----------+-------+-------+-------+
Allocated Heap: unknown
Allocated Stack: unknown
Total Static RAM memory (data + bss): 11080 bytes
Total RAM memory (data + bss + heap + stack): 11080 bytes
Total Flash memory (text + data + misc): 42738 bytes
Image: .build/K64F_ARM_BLINKY/mbed-os-example-blinky.bin
```

Without `--build` or with for example `mbed compile --build .build/K64F/ARM/BLINKY/ -t ARM -m K64F -j0 -c`:

```
Link: mbed-os-example-blinky
Elf2Bin: mbed-os-example-blinky
+---------------------+-------+-------+-------+
| Module              | .text | .data |  .bss |
+---------------------+-------+-------+-------+
| Misc                | 13569 |    20 |  1372 |
| features/frameworks |  4315 |   572 |   296 |
| hal/common          |  3666 |    28 |   168 |
| hal/targets         | 13904 |    76 |   136 |
| rtos/rtos           |   114 |     4 |     0 |
| rtos/rtx            |  6378 |    92 |  8316 |
| Subtotals           | 41946 |   792 | 10288 |
+---------------------+-------+-------+-------+
Allocated Heap: unknown
Allocated Stack: unknown
Total Static RAM memory (data + bss): 11080 bytes
Total RAM memory (data + bss + heap + stack): 11080 bytes
Total Flash memory (text + data + misc): 42738 bytes
Image: .build/K64F/ARM/BLINKY/mbed-os-example-blinky.bin
```

So it seems that when the path doesn't start `.build/<target>/<toolchain>/`, it doesn't work properly.

mbed-cli: 0.9.1
mbed-os: 9111aa4c2dc74fd9bc29f6c37a26489c09cf8a68

sg-, 2016-09-19:
@tommikas Is this resolved?

tommikas, 2016-09-20:
Yep, seems to work fine now. Closing.

closed 2016-09-20 by tommikas