ARMmbed/mbed-os #2626: [STM32F303K8] Serial output wrongly come out from different port

toyowata, 2016-09-06:
### Problem:

By following test code, same output signal from the PB_3 port wrongly come out from the PA_2 port as well.

```
#include "mbed.h"

Serial pc2(PB_3, NC);

int main()
{
    int i = 1;
    pc2.printf("Hello World Serial PORT2 !\n");

    while(1) {
        pc2.printf("PORT2 program runs since %d seconds.\n", i++);
        wait(1.0);
    }
}
```

PB_3 : Serial 2 TX
PA_2 : Serial 2 TX, but not configured by this code

[image]
### Expectation:

There should be no output from the PA_2.

### Version:
The mbed library rev 123 (and onward) had this problem.  **Rev 122 or forward version of mbed library does not have such problem.** i.e. No signal from PA_2 port.

Originally reported [here](https://developer.mbed.org/questions/74082/Nucleo-F303K8Serial-2-TX).

svastm, 2016-09-06:
Hello,

I did not investigate the behaviour change between rev 122 and rev 123 but on this board, STDIO use the USART2 and PA_2 as TX pin. This pin is enabled by mbed to use `printf()`. Also, the 2 pins are connected to the same USART. That's why you see the same output on PA_2 and PB_3.

To correct this, you can disable the `stdio_uart` (if you don't need to use `printf()`)

``` c++
#include "mbed.h"

extern serial_t stdio_uart;
Serial * pc2;

int main()
{
    serial_free(&stdio_uart);

    /* You have to init the new serial after the deinit of the old one */
    pc2 = new Serial(PB_3, NC);

    int i = 1;
    pc2->printf("Hello World Serial PORT2 !\n");

    while(1) {
        pc2->printf("PORT2 program runs since %d seconds.\n", i++);
        wait(1.0);
    }
}
```

Or you can use USART1 instead of USART2.

NB: The definition of the STDIO pins and periph is usually in the `PeripheralNames.h` file.

jeromecoutant, 2017-05-22:
Hi @toyowata 
Is the answer provided by svastm OK for you?
Thx

toyowata, 2017-05-24:
Hi @jeromecoutant @svastm 

The example code from svastm works fine.  However, it is not easy to use the PB_3 port of the 303K8 platform.  If you do not have immediate plan to fix this issue, can you provide additional information for developer in the platform page here?

https://developer.mbed.org/platforms/ST-Nucleo-F303K8/#known-limitations

jeromecoutant, 2017-05-30:
Hi
There is no limitation to add.
In MBED, as soon as target enables DEVICE_SERIAL, a default printf is defined for the console.
Default pins are STDIO_UART_TX and STDIO_UART_RX that you can find in PeripheralNames.h file:
#define STDIO_UART_TX  PA_2
#define STDIO_UART_RX  PA_15

So code from @svastm is freeing first this default printf, before using another one.
Maybe another way of working could be to patch PeripheralNames.h file?

jeromecoutant, 2017-06-21:
May we can close this issue ?
Thx

ST_TO_BE_CLOSED

LMESTM, 2017-06-29:
@toyowata @0xc0170 as explained above, we don't think there is an issue. Could you please either close or provide further explanations ?

closed 2017-06-30 by toyowata