ARMmbed/mbed-os #2628: Please allow any fields in the mbed_app.json file

thegecko, 2016-09-06:
The checks here https://github.com/ARMmbed/mbed-os/blob/master/tools/config.py#L381 enforce that only an `application` entry is allowed in the `mbed_app.json` file. Could this check be removed to allow any entries?

bogdanm, 2016-09-06:
That check doesn't enforce only one `application` entry, but rather checks that only supported entries are present. The supported entries are listed in https://github.com/ARMmbed/mbed-os/blob/master/tools/config.py#L342. If you need other entries, we can modify that list. Allowing any entry doesn't feel like a good idea.

sg-, 2016-09-19:
@thegecko any update on this?

thegecko, 2016-09-19:
I believe @jaustin provided a list of expected keys for this file.

sg-, 2016-09-19:
Then I guess we're just waiting on a PR then.

closed 2017-01-16 by sg-