ARMmbed/mbed-os #2638: InterruptIn rise/fall methods overwrite pullup/pulldown mode for STM32F4 targets

andreaslarssonublox, 2016-09-07:
The gpio_irq_set function calls pin_function with GPIO_NOPULL.

The code below was used to generate the issue on an STM32F429i-DISCO board using the DISCO_F429ZI target. If the mode method is called after rise/fall it works as expected.

`
#include <stdio.h>
#include <string.h>
#include "mbed.h"

InterruptIn button(PF_2);
DigitalOut led1(LED1);
DigitalOut led2(LED2);

void handleRise() {
	led1 = 1;
	led2 = 0;
}

void handleFall() {
	led1 = 0;
	led2 = 1;
}

int main() {
    button.mode(PullUp);      // Set PF_2 to pullup. GPIOF->PUPDR=0x00000010 -> OK
    button.rise(&handleRise); // Will overwrite the mode. GPIOF->PUPDR=0x00000000 -> Not OK
    button.fall(&handleFall); // Will overwrite the mode. GPIOF->PUPDR=0x00000000 -> Not OK
	while (1) {
        wait(0.25);
    } 
}
`

andreaslarssonublox, 2016-09-07:
@bcostm @adustm

LMESTM, 2016-09-07:
hi - I can have a look at it

bcostm, 2016-09-07:
I think if we save the gpio mode inside the gpio_t structure and reuse it after in the gpio_irq_set function it should work...

LMESTM, 2016-09-07:
unfortunately gpio_irq_set is using a gpio_irq_t object, not the gpio_t  one ... the common point being the assigned "pin"  ... I'll look into this further tomorrow

andreaslarssonublox, 2016-09-15:
@LMESTM, any updates on this?

LMESTM, 2016-10-04:
@andreaslarssonublox  I hope you'll agree that this is better late than never ?  ;-)
Anyway I pushed a branch concerning this issue and non regression tests are ongoing locally 
If you have any feedback please let me know

bcostm, 2017-01-10:
I think this issue can be closed.

bridadan, 2017-01-10:
Thanks @bcostm!

closed 2017-01-10 by bridadan