ARMmbed/mbed-os #2655: Export Classic project to Simplicity Studio: cannot find mbed_config.h

janjongboom, 2016-09-09:
When exporting to Simplicity Studio there is no mbed_config.h in the export (or it's not generated by the build tool, either way). Removing the `-include` flag from the build flags create a successful build. As far as I can see mbed_config is not used for any of the EFM32 boards, so perhaps we can just remove that flag from the exporter?

See also https://forums.mbed.com/t/mbed-online-compiler-export-problem/1859

bogdanm, 2016-09-09:
In theory, `mbed_config.h` should be used for every board/target. Removing it is likely to lead to problems, and even worse, they might not be immediately obvious. The proper solution here is to fix the exporter.

0xc0170, 2016-09-12:
cc @stevew817 @akselsm Please have a look

stevew817, 2016-09-12:
We're aware of this issue, and are trying to figure out the best way to fix it.

sg-, 2016-09-19:
@janjongboom is this resolved?

theotherjimmy, 2016-09-19:
But it is related to the PR that mentions this issue, that was merged 3 days ago.

janjongboom, 2016-09-20:
@theotherjimmy Ah, yeah, then probably.

closed 2016-09-20 by sg-