ARMmbed/mbed-os #2660: [LPC1347] PWMout freq on 16 bit timers wrong

JojoS62, 2016-09-10:
### Description
- Type: Bug
- Priority: Major

## Bug
PWMout frequency is higher than set in pwm.period(). This happens on pwm pins with 16 bit counters CT16B0 / CT16B1, on CT32B0 it works as expected.

**Target**
LPC1347

**Toolchain:**
ARM, online compiler

mbed lib 64...125

**Expected behavior**
pwm.period(0.01) -> 10 ms

**Actual behavior**
pwm.period(0.01) -> 0.898 ms

**Steps to reproduce**

```c
#include <mbed.h>

PwmOut pwm_ok(P0_18);
PwmOut pwm_not_ok(P0_9);

int main()
{
    pwm_ok.period(0.01f);       // 10 ms / 100 Hz
    pwm_ok = 0.25f;
    
    pwm_not_ok.period(0.01f);   // 10 ms / 100 Hz
    pwm_not_ok = 0.25f;
    
    while(1);
}
```

[image]

JojoS62, 2016-09-10:
The problem is:  clock of 72 MHz / 2^16 gives a max pwm period time of about 0.9 ms. 
I've used the solution from LPC11XX that sets a prescaler, initial to 31 to get the default setting of 20 ms pwm period time working. For times >28 ms it is necessary to set a higher prescaler value.

0xc0170, 2016-09-12:
cc @toyowata

toyowata, 2016-09-20:
I got the LPCXpresso1347 board and just started to see detail.  I hope same fix as the LPC11XX can be used.  I will investigate more.

JojoS62, 2016-09-20:
Great,

 

And thanks for your help.

I have attached the modified source, I don’t know if my pull request did work.

Another thing to note: there is a definition for CT32B1, but this is used for the internal mbed ticker and also for USB, so these PWM pins should not be used.

 

Then I use the LPC1347 with the LPCXpresso. I have created the startup code for this toolchain and modified the linker script to use the nano lib, otherwise the generated code will not fit into this controller.

 

Regards,

Johannes

 

/* mbed Microcontroller Library
 * Copyright (c) 2006-2013 ARM Limited
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *     http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */
#include "mbed_assert.h"
#include "pwmout_api.h"
#include "cmsis.h"
#include "pinmap.h"

#define TCR_CNT_EN       0x00000001
#define TCR_RESET        0x00000002

/* To have a PWM where we can change both the period and the duty cycle,
 * we need an entire timer. With the following conventions:
 *   * MR3 is used for the PWM period
 *   * MR0, MR1, MR2 are used for the duty cycle
 */
static const PinMap PinMap_PWM[] = {
    /* CT16B0 */
    {P0_8 , PWM_1, 2}, {P1_13, PWM_1, 2},    /* MR0 */
    {P0_9 , PWM_2, 2}, {P1_14, PWM_2, 2},   /* MR1 */
    {P0_10, PWM_3, 3}, {P1_15, PWM_3, 2},   /* MR2 */

    /* CT16B1 */
    {P0_21, PWM_4, 1},                      /* MR0 */
    {P0_22, PWM_5, 2}, {P1_23, PWM_5, 1},   /* MR1 */

    /* CT32B0 */
    {P0_18, PWM_6, 2}, {P1_24, PWM_6, 1},   /* MR0 */
    {P0_19, PWM_7, 2}, {P1_25, PWM_7, 1},   /* MR1 */
    {P0_1 , PWM_8, 2}, {P1_26, PWM_8, 1},   /* MR2 */

    /* CT32B1 */
    {P0_13, PWM_9 , 3}, //{P1_0, PWM_9 , 1},  /* MR0 */
    {P0_14, PWM_10, 3}, //{P1_1, PWM_10, 1},  /* MR1 */
    {P0_15, PWM_11, 3}, //{P1_2, PWM_11, 1},  /* MR2 */

[... 115 lines cut ...]
    }
    timer->TCR = TCR_CNT_EN;
}

void pwmout_pulsewidth(pwmout_t* obj, float seconds) {
    pwmout_pulsewidth_us(obj, seconds * 1000000.0f);
}

void pwmout_pulsewidth_ms(pwmout_t* obj, int ms) {
    pwmout_pulsewidth_us(obj, ms * 1000);
}

void pwmout_pulsewidth_us(pwmout_t* obj, int us) {
    timer_mr tid = pwm_timer_map[obj->pwm];
    LPC_CTxxBx_Type *timer = Timers[tid.timer];
    uint32_t t_on = (uint32_t)((((uint64_t)SystemCoreClock * (uint64_t)us) / (uint64_t)1000000) / (timer->PR + 1)); ;
    
    timer->TCR = TCR_RESET;
    if (t_on > timer->MR3) {
        pwmout_period_us(obj, us);
    }
    uint32_t t_off = timer->MR3 - t_on;
    timer->MR[tid.mr] = t_off;
    timer->TCR = TCR_CNT_EN;
}

toyowata, 2016-09-26:
@JojoS62 Thanks.  I created a PR here: https://github.com/ARMmbed/mbed-os/issues/2810

closed 2016-12-10 by JojoS62