ARMmbed/mbed-os #2667: Clone repos from .lib files to specific directory

daniel-cesarini-tridonic-com, 2016-09-12:
### Description
- Type: Question
- Priority: Major

## Question
**How to instruct ```mbed deploy``` to download/clone repos specifies in .lib files into a specific directory?**

daniel-cesarini-tridonic-com, 2016-09-12:
@MarceloSalazar

javier-moreno-tridonic-com, 2016-09-12:
Just to elaborate a little more, currently we need to use the atmel-rf-driver but just in some targets that include the atmel chip. We are including it using a .lib file. Since deploying puts all the libraries referenced by lib files into the root, the system tries to compile it anyway.

Since some of our targets do not use the FEATURE_IPV6 the compilation fails due to the nanostack not being present in the build path.

Our first idea was a @daniel-cesarini-tridonic-com suggest to be able to pull the code into a folder that we could then reference with target labels.

We are also open to alternatives to solve our issue.

sg-, 2016-09-12:
@daniel-cesarini-tridonic-com I think this patch will solve your problem. https://github.com/ARMmbed/mbed-os/pull/2551 Would you pull it locally and have a try?

daniel-cesarini-tridonic-com, 2016-09-13:
@sg- thanks for the answer. I think that I did not elaborate well enough what we are looking for:
I am looking for a mechanism to have ```atmel-rf-driver.lib``` and other custom ```module_a.lib``` cloning the  repos to specific folders (e.g. to ./FEATURES/TARGET_MODULE_A ), such that they can be compiled or not depending on the targets.json configuration.

@javier-moreno-tridonic-com: I think we already did something in the direction of de-coupling from the atmel-rf chip. Is https://github.com/ARMmbed/mbed-os/pull/2551 still relevant for us? But this is then going out of scope from this issue.

sg-, 2016-09-13:
> I am looking for a mechanism to have atmel-rf-driver.lib and other custom module_a.lib cloning the repos to specific folders (e.g. to ./FEATURES/TARGET_MODULE_A ), such that they can be compiled or not depending on the targets.json configuration.

The only way to do this is to protect with a macro that evaluates to ```#if 0``` using macros add and remove. mbed deploy will always download all *.lib repos onto disc.

ivan-adzic-tridonic-com, 2016-09-13:
> mbed deploy will always download all *.lib repos onto disc.

@sg- We are fine with that, just wondering if it is possible to instruct somehow location where it will be downloaded, that it is not the root path always, like @daniel-cesarini-tridonic-com 

>...cloning the repos to specific folders?

sg-, 2016-09-13:
> just wondering if it is possible to instruct somehow location where it will be downloaded,

Anywhere in the tree it finds the .lib file. Just move this where you want it.

ivan-adzic-tridonic-com, 2016-09-13:
Thank you for support! it works! :+1:

daniel-cesarini-tridonic-com, 2016-09-13:
Thanks. :+1:

closed 2016-09-13 by daniel-cesarini-tridonic-com