ARMmbed/mbed-os #2668: Binary size changes between two consecutive builds

mottigondabi, 2016-09-12:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**Toolchain version:**
4.9.3 20150529

**mbed-cli version:** 
0.8.8

**mbed-os sha:**
d1f6cb4 Updating mbed-os to mbed-os-5.1.3

**Steps to reproduce**
Made the following:
1. cloned the blinky example - git@github.com:ARMmbed/mbed-os-example-blinky.git
2. build according to the given README instructions
3. build again with the "-c" (clean) switch

**Expected behavior**
*mbed-os-example-blinky.bin*:
 - Bullet (2) yield a binary size of 56912 bytes
 - Bullet (3) yield a binary size of **56912** bytes

**Actual behavior**
*mbed-os-example-blinky.bin*:
 - Bullet (2) yield a binary size of 56912 bytes
 - Bullet (3) yield a binary size of **56920** bytes

## Question
I expected both builds (2) & (3) should yield same resulting binary size. Any idea why there is a deviation of 8 bytes?

This behavior observed in our provisioning build system as well and I happen to be using the blinky example app to narrow down this issue to simplify reproduction.

@sg- 
@jenia81 
@trianglee

sg-, 2016-09-12:
This could be the same as addressed in https://github.com/ARMmbed/mbed-os/pull/2604 and reported https://github.com/ARMmbed/mbed-cli/issues/323

sg-, 2016-09-19:
@mottigondabi is this resoled?

theotherjimmy, 2017-04-25:
I can no longer reproduce this issue. @mottigondabi Feel free to re-open if it's still a problem for you.

closed 2017-04-25 by theotherjimmy