ARMmbed/mbed-os #2670: mbed 2.0 - NUCLEO-F303K8 & L432KC bug in Serial#attach function

janjongboom, 2016-09-12:
See https://developer.mbed.org/questions/74236/Possible-bug-in-Serial-attach-function-w/

> I think mbed 2.0 Rev 125 introduced a bug in the attach() function of the Serial class for the NUCLEO-F303K8 target. The attach() function behaves as expected for the NUCLEO-F303K8 using mbed 2.0 Rev 124 .
>
> The same bug also seems to appear in all versions of the mbed 2.0 for the NUCLEO-L432KC target, which was first released in Rev 122.
>
> My application uses Serial1 to read packet data from the serial output of an r/c receiver using interrupts. It uses the default Serial2 as a monitor. For safety and other reasons, the r/c transmitter is typically powered up and transmitting data prior to powering up the r/c receiver. This means that I want to be able to initialize the NUCLEO’s Serial1 and attach the interrupt with serial data present at the Rx pin of Serial1.

Confirmed by others in the thread that this is a problem on a number of NUCLEO boards in rev. 125, but not in rev. 124.

LMESTM, 2017-06-29:
@janjongboom 
Hey - sorry for not supporting earlier ... is the problem still visible on recent MBED versions ? 
I tried to reproduce here with MBED2 Serial interrupt test (PC sends data to target ..) and could not reproduce. If you have another sample code I can try again

janjongboom, 2017-07-12:
Issue solution seems to be verified in rev145, as per the comment on mbed.org.

closed 2017-07-12 by janjongboom