ARMmbed/mbed-os #2673: TCPSocket fails after receiving 140 KiB cumulative

marcuschangarm, 2016-09-12:
After downloading 140 KiB of data, TCP::recv returns -3011. All further calls to the same socket does nothing.

Test application: https://github.com/ARMmbed/test-mbed-tcpsocket

This issue might be related to https://github.com/ARMmbed/mbed-os/issues/2553

Tested on K64F and GCC_ARM.

sg-, 2016-09-13:
@geky

marcuschangarm, 2016-09-15:
Tested on DISCO_F429ZI with the same result.

geky, 2016-09-15:
Manually inspection reveals lwip is returning ERR_RST. This is actually related to https://github.com/ARMmbed/mbed-os/issues/2554.

However, even with that fix, the test program fails to fully download the file (terminates after ~140KiB of data). It looks like there is a timeout on the server side that the board is failing to meet. This script exhibits the same problem locally:
``` python
from socket import *
import time

header = (
    "GET /lootbox/firmware/alice.txt HTTP/1.1\r\n"
    "Host: s3-us-west-2.amazonaws.com\r\n"
    "Range: bytes=%d-%d\r\n\r\n")

s = socket(AF_INET, SOCKET_STREAM)
s.connect(("s3-us-west-2.amazonaws.com", 80))

for i in range(0, 200000, 1000):
    s.send(header % (i, i+1000))
    dat = s.recv(2048)
    if not dat:
        break

    print(repr(dat[:24]))
    print(header % (i, i+1000))
    time.sleep(0.5)
```

It looks like this is a defensive feature of Apache:
https://en.wikipedia.org/wiki/HTTP_persistent_connection

To properly handle long-running persistent downloads, you will need to handle the disconnect case and reconnect the socket to resume downloading the payload with the `Range` field.

marcuschangarm, 2016-09-16:
Hi @geky !

Thank you for the detailed investigation! I wasn't aware of the aggressive HTTP timeout. 

The unresponsive calls I was seeing after an error was due to an improperly handled event after closing the socket!

closed 2016-09-16 by marcuschangarm