ARMmbed/mbed-os #2676: mbed 2: Test MBED_A28

sg-, 2016-09-13:
### Description
mbed 2 test failure on MBED_A28

```
GCC
	
Building project test (NUCLEO_F334R8, GCC_ARM)
Scan: can_loopback
Scan: mbed
Scan: env
Compile: main.cpp
[Error] main.cpp@24,14: expected constructor, destructor, or type conversion before '(' token

ARM, uARM
	
Building project test (NUCLEO_F334R8, uARM)
Scan: can_loopback
Scan: mbed
Scan: env
Compile: main.cpp
[Error] main.cpp@24,0:  #77-D: this declaration has no storage class or type specifier
[Error] main.cpp@24,0:  #20: identifier "TARGET_NUCLEO_F446ZE" is undefined
[Error] main.cpp@24,0:  #65: expected a ";"
[Warning] main.cpp@29,0:  #12-D: parsing restarts here after previous syntax error

IAR
	
Building project test (NUCLEO_F334R8, IAR)
Scan: can_loopback
Scan: mbed
Scan: env
Compile: main.cpp
[Error] main.cpp@24,16: [Pe077]: this declaration has no storage class or type specifier
[Error] main.cpp@24,38: [Pe020]: identifier "TARGET_NUCLEO_F446ZE" is undefined
[Error] main.cpp@24,22: [Pe065]: expected a ";"
```

---------------------------------------------------------------
**Target**
NUCLEO_F334R8

**Toolchain:**
GCC_ARM, ARM, uARM, IAR

jeromecoutant, 2016-09-13:
With mbed_lib_rev125, MBED_A28 result is OK with GCC_ARM, ARM, uARM, IAR for NUCLEO_F334R8

sg-, 2016-09-13:
This test was against master

jeromecoutant, 2016-09-13:
Yes, there was a error with last commit d0e2f1efa6de02e1d522a025995d16f7b1de9edd

a #elif has been deleted....

jeromecoutant, 2016-09-13:
This should be corrected with https://github.com/ARMmbed/mbed-os/pull/2682
Thx @ohagendorf

adustm, 2017-01-03:
Hello @sg- , happy 2017 to you :)

I think this issue is fixed now. Could you close it ? 
Cheers
Armelle

closed 2017-01-03 by 0xc0170

sg-, 2017-01-06:
@adustm Happy new year!