ARMmbed/mbed-os #2677: mbed 2: RTOS Enabled ST-Nucleo_L053R8

sg-, 2016-09-13:
### Description
- Type: Bug

Link errors when compiling RTOS tests for NUCLEO_L053R8 https://developer.mbed.org/platforms/ST-Nucleo-L053R8/#known-limitations But there is support in the codebase causing CI compile failures with ARM and IAR compiler for mbed 2 builds.
---------------------------------------------------------------
## Bug

**Target**
NUCLEO_L053R8

**Toolchain:**
ARM and IAR

jeromecoutant, 2016-09-13:
Hi

in https://github.com/ARMmbed/mbed-os/pull/2648
I have removed rtos compilation from travis script

Regards,

sg-, 2016-09-13:
Its more than travis, its also tests.py. If there are no tests I'd suggest RTOS is turned off for the target.

jeromecoutant, 2016-09-13:
yes, merge https://github.com/ARMmbed/mbed-os/pull/2648 :-)

bcostm, 2016-11-18:
I have generated a project using `mbed export`  and I still have linking error with uvision5:
```
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching rt_task.o(.bss).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching us_ticker.o(.bss).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching hal_tick.o(.bss).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching rt_list.o(.bss).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching rtx_conf_cm.o(.bss).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching rtx_conf_cm.o(.bss).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching filebase.o(.bss).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching rtx_conf_cm.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching main.o(.bss).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching rtx_conf_cm.o(.bss).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching rtx_conf_cm.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching rt_cmsis.o(.bss).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching rt_cmsis.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching us_ticker.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching mbed_critical.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching serial_api.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching mbed_us_ticker_api.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching rt_oseventobserver.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching rt_robin.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching rt_system.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching rt_task.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching retarget.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching retarget.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching rtx_conf_cm.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching rtx_conf_cm.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching rtx_conf_cm.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching rt_time.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching rtos_idle.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching thread.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching stm32l0xx_hal.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching hal_tick.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching system_stm32l0xx.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching filebase.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching stdio_streams.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching stdio_streams.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching stdio_streams.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6406E: No space in execution regions with .ANY selector matching streamlock.o(.data).
.\.build\uvision5\mbed-os.axf: Error: L6407E: Sections of aggregate size 0x224 bytes could not fit into .ANY selector(s).
```
It looks like the RTOS files are still included during the link ?

I have well checked the `One ELF Section per Function` option in the C/C++ settings and set option `-O3`.

0xc0170, 2016-11-18:
@bcostm yes, using mbed export you are using mbed-os 5 , thus rtos is included by default. That target should not be possible to export .. if it's not supported for mbed 5 (why the export does not fail? @theotherjimmy )

sg-, 2017-01-19:
@theotherjimmy bump regarding the export question!

theotherjimmy, 2017-01-19:
`mbed export` will generate an mbed-os 5 style project. To export a mbed 2 project is a bit more involved...

theotherjimmy, 2017-01-19:
@0xc0170 The export does not for a complex reason: mbed-os 5 vs 2 builds are actually treated the same by the tools; the real difference is in how mbed is included. It becomes especially hard, because of the aforementioned fact, to distinguish between 5 vs. 2 projects at compile/export time. So we don't. We could, if wanted too, probably.

0xc0170, 2017-01-19:
Thanks @theotherjimmy for the explanation. I am closing this as it was resolved above via 2 PR referenced in this issue.

closed 2017-01-19 by 0xc0170

theotherjimmy, 2017-01-19:
@0xc0170 If you think that trying to distinguish at export time would help, We can open an issue to track that feature request/bug report.