ARMmbed/mbed-os #2689: NUCLEO_L053R8 and NUCLEO_F334R8	- build failures (

0xc0170, 2016-09-13:
### Description
- Type: Bug |
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_L053R8, NUCLEO_F334R8

**Toolchain:**
depends on the test 

** Tests **

NUCLEO_L053R8
```
// ARM
Timer, Mail messaging and Queue messaginig fail

// IAR
all rtos fail to build
```

NUCLEO_F334R8
```
MBED_A28 fails for all toolchains

Compile: main.cpp
[Error] main.cpp@24,0:  #77-D: this declaration has no storage class or type specifier
[Error] main.cpp@24,0:  #20: identifier "TARGET_NUCLEO_F446ZE" is undefined
[Error] main.cpp@24,0:  #65: expected a ";"
[Warning] main.cpp@29,0:  #12-D: parsing restarts here after previous syntax error
```

@bcostm @adustm @LMESTM

ohagendorf, 2016-09-13:
for the MBED_A28 failure please see PR #2682

jeromecoutant, 2016-09-13:
See https://github.com/ARMmbed/mbed-os/pull/2682
and https://github.com/ARMmbed/mbed-os/pull/2648
Thx

0xc0170, 2016-09-13:
Thank you guys for referencing , we will go through all those, and run tests

adustm, 2017-01-03:
Hello, 
with mbed_lib_rev132, we have : 
* every build ok for NUCLEO_L053RB, 
* the following build ok / nok for NUCLEO_F334R8
```
Build successes:
  * NUCLEO_F334R8::ARM::MBED_A28
  * NUCLEO_F334R8::ARM::RTOS
  * NUCLEO_F334R8::ARM::RTOS_1
  * NUCLEO_F334R8::ARM::RTOS_2
  * NUCLEO_F334R8::ARM::RTOS_3
  * NUCLEO_F334R8::ARM::RTOS_4
  * NUCLEO_F334R8::ARM::RTOS_5
  * NUCLEO_F334R8::ARM::RTOS_6
  * NUCLEO_F334R8::ARM::RTOS_7
  * NUCLEO_F334R8::ARM::RTOS_8
  * NUCLEO_F334R8::GCC_ARM::MBED_A28
  * NUCLEO_F334R8::GCC_ARM::RTOS
  * NUCLEO_F334R8::IAR::MBED_A28
  * NUCLEO_F334R8::IAR::RTOS
  * NUCLEO_F334R8::IAR::RTOS_1
  * NUCLEO_F334R8::IAR::RTOS_2
  * NUCLEO_F334R8::IAR::RTOS_3
  * NUCLEO_F334R8::IAR::RTOS_4
  * NUCLEO_F334R8::IAR::RTOS_5
  * NUCLEO_F334R8::IAR::RTOS_6
  * NUCLEO_F334R8::IAR::RTOS_7
  * NUCLEO_F334R8::IAR::RTOS_8

build failed for
  * NUCLEO_F334R8::GCC_ARM::RTOS_1
  * NUCLEO_F334R8::GCC_ARM::RTOS_2
  * NUCLEO_F334R8::GCC_ARM::RTOS_3
  * NUCLEO_F334R8::GCC_ARM::RTOS_4
  * NUCLEO_F334R8::GCC_ARM::RTOS_5
  * NUCLEO_F334R8::GCC_ARM::RTOS_6
  * NUCLEO_F334R8::GCC_ARM::RTOS_7
  * NUCLEO_F334R8::GCC_ARM::RTOS_8
  * NUCLEO_F334R8::uARM::RTOS_1
  * NUCLEO_F334R8::uARM::RTOS_2
  * NUCLEO_F334R8::uARM::RTOS_3
  * NUCLEO_F334R8::uARM::RTOS_4
  * NUCLEO_F334R8::uARM::RTOS_5
  * NUCLEO_F334R8::uARM::RTOS_6
  * NUCLEO_F334R8::uARM::RTOS_7
  * NUCLEO_F334R8::uARM::RTOS_8
```

@jeromecoutant are you aware of this ?
Cheers
Armelle

jeromecoutant, 2017-02-21:
Hi
Can we close this issue ?

NUCLEO_L053R8
- Target doesn't support RTOS as RAM size is not sufficient

NUCLEO_F334R8:
- MBED_A28 is OK for ARM/uARM/GCC/IAR since mbed_lib_rev127
- Target doesn't support RTOS for uARM and GCC as small lib is used

bcostm, 2017-04-07:
TO_BE_CLOSED

closed 2017-04-07 by 0xc0170