ARMmbed/mbed-os #2690: Exporters - mbed export -c removes git repository

0xc0170, 2016-09-13:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
Any

**Toolchain:**
Any

**Toolchain version:**

**mbed-cli version:** 
0.9.1

**meed-os sha:**
dd5649d92b2d292cd94e0a784cb0f2b65b3f2e84

**Steps to reproduce**

Export a simple project like blinky, or don't probably need to even export. Run ``mbed export -c -i uvision`` , and your .git folder disappears and you get errors that your example is not git repository anymore.

```
 mbed export -c --i uvision
Traceback (most recent call last):
  File "C:\2\mbed-os-example-blinky\mbed-os\tools\project.py", line 227, in <module>
    main()
  File "C:\2\mbed-os-example-blinky\mbed-os\tools\project.py", line 223, in main
    zip_proj=zip_proj, options=options.opts)
  File "C:\2\mbed-os-example-blinky\mbed-os\tools\project.py", line 87, in export
    macros=macros, libraries_paths=lib, zip_proj=zip_name, options=options)
  File "C:\2\mbed-os-example-blinky\mbed-os\tools\project_api.py", line 181, in export_project
    rmtree(export_path)
  File "C:\Python27\lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Python27\lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Python27\lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Python27\lib\shutil.py", line 252, in rmtree
    onerror(os.remove, fullname, sys.exc_info())
  File "C:\Python27\lib\shutil.py", line 250, in rmtree
    os.remove(fullname)
WindowsError: [Error 5] Access is denied: '.\\.git\\objects\\pack\\pack-ba9f2748f01b8fc2786f100f74fe30b7d905f4a6.idx'
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\2\mbed-os-example-blinky\mbed-os\tools\project.py -i uvision -m UBLOX_C029 -c --source ." in "C:\2\mbed-os-example-blinky"
```

@theotherjimmy @sarahmarshy

theotherjimmy, 2016-09-13:
Looking into it now. Thanks for the bug report @0xc0170 !

theotherjimmy, 2016-09-13:
Reproduced. Kills everything, not just the git repo.

theotherjimmy, 2016-09-13:
The problem is that we are cleaning in two places, one within `project.py` and the other within `project_api.py`. Given that the `export_path` may be the same as the project path, and that the `tools.paths.EXPORT_DIR` is never equal to the project path, we should probably be 'cleaning' the latter. Further, I'm not sure what a user would expect the clean flag to do.

theotherjimmy, 2016-09-13:
I just tried switching out all of the cleaning for git clean -fe mbed-os, and I have to say it's pretty nice.

eg:
```
Removing Makefile
Removing mbed-os-example-blinky.uvopt
Removing mbed-os-example-blinky.uvproj
Removing mbed_config.h
Build Options: debug-info
Scan: .
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_UVISOR
Scan: FEATURE_BLE
Scan: FEATURE_CLIENT
Scan: FEATURE_IPV6
Scan: FEATURE_IPV4
Scan: FEATURE_STORAGE
```

theotherjimmy, 2016-09-13:
I'll put up a PR, and we can discuss alternate implementations there.

theotherjimmy, 2016-09-13:
https://github.com/ARMmbed/mbed-os/pull/2692 PR up.

0xc0170, 2016-09-14:
Thanks @theotherjimmy for looking, I'll have a look at that PR

theotherjimmy, 2016-09-14:
I'm thinking that doing a clean before exporting will be an mbed-cli issue in the future. I might just re-open that PR with the first commit in the series.

0xc0170, 2016-09-16:
@theotherjimmy What's the status ? I noticed that PR was closed.

theotherjimmy, 2016-09-16:
>I'm thinking that doing a clean before exporting will be an mbed-cli issue in the future. I might just re-open that PR with the first commit in the series.

closed 2016-12-23 by 0xc0170