ARMmbed/mbed-os #2693: AnalogOut not working on K64F

BlackstoneEngineering, 2016-09-13:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM, ARMv5

**Toolchain version:**
ARM_Compiler_5.06u3
arm-none-eabi-gcc  4.9.3 20141119

**mbed-cli version:** 
(`mbed --version`)
0.9.1

**meed-os sha:**
(`git log -n1 --oneline`)
61bf14d updated mbed-os library to current revision

**DAPLink version:**

**Expected behavior**
`AnalogOut aout(aout_pin);` should allow you to initialize an analog out

**Actual behavior**
`AnalogOut aout(aout_pin);` hangs and never completes. In greentea tests they always timeout. When run on the LPC1768 it works just fine, which makes me think its a fault of the K64F underlying port. 

**Steps to reproduce**
pull down the CI test shield tests, run `mbed test -n tests-api-analogout`.
https://github.com/ARMmbed/ci-test-shield/blob/master/TESTS/API/AnalogOut/AnalogOut.cpp#L15

BlackstoneEngineering, 2016-09-13:
@maclobdell here

0xc0170, 2016-09-14:
cc @mmahadevan108

mmahadevan108, 2016-09-15:
The below link is not working for me:
https://github.com/ARMmbed/ci-test-shield/blob/master/TESTS/API/AnalogOut/AnalogOut.cpp#L15

Can you please double check and resend.

BlackstoneEngineering, 2016-09-15:
Right, thats cause that repo is not public yet.

Issue has been resolved, I misread the pinout diagram as AnalogOut being on A5, when in fact it is on the internal header next to A5. Fly wired in correct pin and it passes the tests. 

This is where I do an epic face palm and appologize profusely.

Closing issue .

closed 2016-09-15 by BlackstoneEngineering

mmahadevan108, 2016-09-15:
Glad to note it all worked out for you.