ARMmbed/mbed-os #2706: Does the `.gitignore` file for mbed-os need to have `lib` in it?

RobMeades, 2016-09-14:
### Description
- Type: Question
- Priority: Minor

-----------------------------------------------------------------
## Question

The `.gitignore` file that arrives with mbed has `lib` in its list.  Our mbed platform has a set of files, in a tree that is provided to us, with a sub-directory named `lib` ('cos those files are considered part of a library).  So this is a real pain for us: git entirely ignores changes in somewhat important files.  Is it really necessary to have 'lib' in the list?

bridadan, 2016-09-14:
I would think it wouldn't be necessary, but I may be missing something here.

Any thoughts @screamerbg ?

screamerbg, 2016-09-14:
Don't think that both lib and lib64 should be there.

bridadan, 2016-09-14:
@RobMeades Would you mind submitting a PR with those changes? If someone else thinks this will cause issues they can bring it up there.

closed 2016-09-19 by sg-