ARMmbed/mbed-os #2711: After exporting and compiling with IAR IDE, can't compile with 'mbed compile' (duplicate definitions)

maclobdell, 2016-09-14:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue: N/A
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
N/A

**Toolchain:**
IAR (possibly others also)

**Toolchain version:**
IAR 7.40

**mbed-cli version:** 
0.8.9

**meed-os sha:**
7669d7f8f5669dc02a8b38747a8229b22c8b4ea0

**DAPLink version:**
N/A

**Expected behavior**
After exporting to IAR IDE, I expect to be able to compile in IAR IDE, and still be able to go back to compiling on the command line, with IAR, ARM, or GCC_ARM, etc.  

**Actual behavior**
After exporting to IAR IDE, the project files are created in the root project folder.  After compiling with IAR IDE, IAR will create a 'build' folder.  If you go back to the command line and try to compile, the linker will find objects in the 'build' folder, try to link them in.  This causes duplicate definitions error with IAR on command line, or a crash with GCC_ARM on command line.   

**Steps to reproduce**

mbed export -m K64F -i IAR

Open project workspace in IAR, compile.

Go back to command line:

mbed compile -m K64F -t IAR
.
.
.
Error[Li006]: duplicate definitions for "_alloc_box"; in "C:\Users\maclob01\Doc
          uments\morpheus\onsemi\porting4\.build\NCS36510\IAR\mbed-os\rtos\rtx\
          TARGET_CORTEX_M\TARGET_M3\TOOLCHAIN_IAR\HAL_CM3.o", and "C:\Users\mac
          lob01\Documents\morpheus\onsemi\porting4\build\iar_arm\Obj\HAL_CM3.o"
Error[Li006]: duplicate definitions for "_free_box"; in "C:\Users\maclob01\Docu
          ments\morpheus\onsemi\porting4\.build\NCS36510\IAR\mbed-os\rtos\rtx\T
          ARGET_CORTEX_M\TARGET_M3\TOOLCHAIN_IAR\HAL_CM3.o", and "C:\Users\macl
          ob01\Documents\morpheus\onsemi\porting4\build\iar_arm\Obj\HAL_CM3.o"
Fatal error[Ms013]: Error limit reached
Fatal error detected, aborting.

maclobdell, 2016-09-14:
@theotherjimmy @screamerbg

theotherjimmy, 2016-09-19:
@sg- agreed. It seems to be using a folder called `Debug` for it's object files, library files, and elf files. This is defined in the template here: https://github.com/ARMmbed/mbed-os/blob/master/tools/export/iar_template.ewp.tmpl#L18-#L29

theotherjimmy, 2016-12-20:
This appears to be fixed on master. The exported IAR project puts everthing into `BUILD` which is the same as used by the build system.

closed 2016-12-20 by theotherjimmy