ARMmbed/mbed-os #2718: Header file for targets.json

tibmanus, 2016-09-15:
## Enhancement

**Reason to enhance or problem with existing solution**
I didn't find a header file generated from the targets.json as there is a similar one to mbed_config.json. This would do a great help for preprocessing functions in IDEs.

0xc0170, 2016-09-16:
> I didn't find a header file generated from the targets.json as there is a similar one to mbed_config.json

Can you please elaborate? What would that header file contain?

tibmanus, 2016-09-16:
I am especially interested in the device_has field in the json, as I see these are generated into defines during compilation.
Example: InterruptIn.h is guarded by an `#if DEVICE_INTERRUPTIN`, so Eclipse won't see it during preprocessing.

bogdanm, 2016-10-27:
I don't know how code completion in Eclipse works, so would it help if we generated a header file called `device_config.h` with all the required `#define DEVICE_something` in it? This file shouldn't be included by any ohter file in the project (since all the required `DEVICE_something` defines are already specified on the compiler's command line and including a file that defines them again might trigger a lot of warnings from the compiler). Would this work?

tibmanus, 2016-11-05:
Yes, that would serve its purpose perfectly.

theotherjimmy, 2016-12-20:
I could add this to the eclipse exporter if that would help you guys out. Would you like to see the generation of this header file in the eclipse_* exporters?

theotherjimmy, 2017-04-25:
@tibmanus bump. Could you answer my question?

tibmanus, 2017-05-03:
@theotherjimmy Yes please.

theotherjimmy, 2017-05-03:
Cool. That should be pretty simple to do. I'll schedule this one then.

theotherjimmy, 2017-05-17:
Resolved by the addition of the `eclipse_<compiler>` and `gnuarmeclipse` exporters.

closed 2017-05-17 by theotherjimmy