ARMmbed/mbed-os #2724: How to use Socket attach()

tibmanus, 2016-09-16:
### Description
- Type: Enhancement | Question

---------------------------------------------------------------

## Question

In mbed OS 3 the network Socket API has a callback when a message is sent on the interface. In mbed  OS 5 there is an attach() that can 

> Register a callback on state change of the socket. The specified callback will be called on state changes such as when the socket can recv/send/accept successfully and on when an error  occurs. The callback may also be called spuriously without reason.

**How to use the attach()?** Since the callback can be called with no reason, it is useless without knowing what happened. In the nanostack socket_api.h there is a list of socket events (together with SOCKET_TX_DONE), but as I see most of the events are just calling the callback passed in attach().

I would appreciate a passed in argument with the socket event, or a similar API as in mbed OS 3.

0xc0170, 2016-09-16:
cc @geky

geky, 2016-09-29:
Hi @tibmanus, the initial implementation of the asynchronous API for sockets was designed to be easy to implement to streamline adding new devices.

The attach function can still be used asynchronously without knowing which exact event occured, by trying the required operation and checking for blocking-failures:
``` cpp
TCPSocket socket(&iface);
Semaphore socket_sema;

void socket_callback() {
    socket_sema.release();
}

int main() {
    socket.connect("8.8.8.8", 80);
    socket.set_blocking(false);
    socket.attach(socket_callback);

    while (true) {
        int err;
        do {
            err = socket.send(data, sizeof(data));
        } while (err != NSAPI_ERROR_WOULD_BLOCK);

        // do other things

        socket_sema.wait();
    }
}
```

We are currently looking into reintroducing an async api similar to mbed OS 3, that matches the async api present in the hardware drivers, but this is unfortunately not ready yet:
https://github.com/ARMmbed/mbed-os/blob/master/hal/api/SerialBase.h#L95-L100

closed 2017-01-18 by sg-