ARMmbed/mbed-os #2732: Tools: synch.py needs changes for updating Mercurial source code repos

adbridge, 2016-09-16:
### Description

When running synch.py to update source files in mbed2 Mercurial repos, there is a whole list of repos which are attempted to be updated:
"mbed-dev"  
"mbed-rtos"
"mbed-dsp"
"mbed-rpc"
"lwip"
"lwip-sys"
"Socket"
"lwip-eth"
"EthernetInterface"
"USBDevice"
"USBHost"
"CellularModem"
"CellularUSBModem"
"UbloxUSBModem"
"UbloxModemHTTPClientTest"
"UbloxModemSMSTest"
"FATFileSystem"

We only need to be updating 
"mbed-dev"  
"mbed-rtos"

Once the source files have been updated the user is given the option to 
commit and push ? Y/N ?
These should be separated. Commiting should be the question asked. Pushing should be left to the user to do separately. This allows the user to push to a private fork in order to test compile before committing to mbed-official.

The Y/N response is also case sensitive which it shouldn't be.

- Type: Bug
- Priority: Major

bridadan, 2017-01-12:
@adbridge Is there an update on this issue?

adbridge, 2017-01-13:
Fix is in progress now

theotherjimmy, 2017-04-25:
@adbridge What was the PR for that fix?

adbridge, 2017-04-26:
Fixed by https://github.com/ARMmbed/mbed-os/pull/3689

closed 2017-04-26 by adbridge