ARMmbed/mbed-os #2736: Project generator for Em::Blocks: Duplicated macros and zero optimization

ivankravets, 2016-09-18:
Hi,

1. There is a lot of duplicated macros in output `EIX` (DEVICE_SPI=1, etc...)
2. Project generator generates `-O0` by default. Is this a bug or feature? The simple project for NUCLEO_F030R8 with "mbed serial" in a few lines errors this message:
```
section `.text' will not fit in region `FLASH'
```

`Os` resolves it.

Thanks!

0xc0170, 2016-09-19:
> Project generator generates -O0 by default. Is this a bug or feature? The simple project for NUCLEO_F030R8 with "mbed serial" in a few lines errors this message:

Should not be ``-O0`` by default, however many of exporters do it from my experience. It should follow the flags that are set for the command line.

cc @EmBlocks - please have a look

theotherjimmy, 2016-09-19:
The macro duplication can simply be removed. The exporters default to using the `-o debug-info` switches. This can be changed by adding a `-o` to the command line of the export command.

Testato, 2016-09-20:
Also on EmBitz forum we are talking about this:
http://www.emblocks.org/forum/viewtopic.php?f=29&t=774#p4310

theotherjimmy, 2016-09-20:
I read the forum link. I have another suggestion: Use the makefile exporter coupled with a simple project file that points to the Makefile.

theotherjimmy, 2017-01-03:
@ivankravets Since you seem interested in the Em::Blocks/EmBitz exporter, could you review #3521 ?

ivankravets, 2017-01-03:
@theotherjimmy thanks! I've just tested it and I don't see any duplicates. 👍

PR #3521 (2017-01-09): Repair the Emblocks exporer and rename to EmBitz
## Description
 * Renamed because the tool has been renamed
 * Removed duplicate symbol definitions
 * Removed common flags from ld flags (common are c/c++ only)
 * Moved EmBlitz to it's own folder

## Status
**READY**

## Migrations
Exporter name changed. No one needs to 'migrate' though.

## Related Issues
Resolves #2912 
Resolves #2736 

## Todos
- [x] Add Reviewers
- [x] Reviews
- [x] Manually check that a few exports build

## Steps to test or reproduce
`mbed export -i embitz -m ...`
closed 2017-01-09 by sg-