ARMmbed/mbed-os #2737: NetworkInterface does not support get_mac_address()

teetak01, 2016-09-19:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM|ARM|IAR

**meed-os sha:**
latest release/master

Currently I cannot call get_mac_address() with NetworkInterface class. This could be fixed by adding:

 virtual const char *get_mac_address() = 0;

to features/net/network-socket/NetworkInterface.h

Could this be added there?

geky, 2016-09-21:
The `get_mac_address` function has actually been moved as a part of https://github.com/ARMmbed/mbed-os/pull/2664, which is just awaiting review/merge.

jupe, 2016-10-06:
seems that PR is merged and there is get mac function available: 
https://github.com/ARMmbed/mbed-os/pull/2664/files#diff-23aab6a3d3cb526bb09cb1e55eb0525dR21

jupe, 2016-10-06:
perhaps this is fixed @teetak01 ?

teetak01, 2016-10-25:
Nice, missed this update. Its working now. Thanks @geky @jupe

closed 2016-10-25 by teetak01