ARMmbed/mbed-os #2738: Please consider generating header with device-specific configuration

yankee14, 2016-09-19:
## Enhancement

**Reason to enhance or problem with existing solution**
I recently ran into an issue where my IDE would recognize certain common classes like DigitalOut or DigitalIn, but would not recognize device-specific classes such as PwmOut. See my [stackoverflow post](https://stackoverflow.com/questions/39553139/why-does-eclipses-indexer-recognize-some-classes-but-not-others) for more detail.

Basically, in order to resolve the issue, I had to replace the device.h that came with my developer.mbed.org exported project with the one located at the [mbed repo here](https://developer.mbed.org/users/mbed_official/code/mbed-src/file/a11c0372f0ba/targets/hal/TARGET_NXP/TARGET_LPC176X/TARGET_MBED_LPC1768/device.h). After that, my IDE had no trouble eating device-specific classes like PwmOut.

ARM employee? Martin Kojtal aka. @0xc0170 suggested I ask about this improvement here.

**Suggested enhancement**
Generate a header file with device-specific configuration for each mbed controller.

**Pros**
I just learned what a header file is a week ago, so I doubt I'm qualified to be listing the pros and cons of something like this. However, I think people who want to setup an offline toolchain with an IDE of some sort will have a much easier time if you include a header with #defines relevant to the controller type they set on the wonderful <sup><sup>\**coughs*\*</sup></sup> online compiler. :D

**Cons**
I'm not sure.

tibmanus, 2016-09-19:
Ref #2718

bogdanm, 2016-10-27:
@yankee14, I read your description of the problem on stackoverflow.com, but I still don't understand why Eclipes behaves like that. The way I see it, DigitalIn/DigitalOut should have exactly the same "visibility" as far as Eclipse is concerned. `device.h` is now obsolete in mbed, we replaced it with data read from `targets.json`. We could generate a file with a different name (but similar content with `device.h`) if that helps Eclipse, but first we need to understand how the code completion works (why it works fine for some classes, but not for others).

bogdanm, 2016-10-27:
This is a duplicate of #2718. let's continue the discussion there.

closed 2016-10-27 by bogdanm