ARMmbed/mbed-os #2742: Automatically switch targets

andcor02, 2016-09-19:
### Automatic switch of target when importing new program
- Type: Enhancement
- Priority: Minor
----------------------------------------------------------------
## Enhancement

So currently, when a new user imports a program for specific platform the online IDE does not automatically switch to that target. While this may not be an issue to a lot of people if we think about the user case of the micro:bit i have had reports of people not understanding why the issue they receive is due to having a different platform selected.

Seems a silly issue to have when this platform should be as streamlined as possible.

Thoughts?

sg-, 2016-09-19:
There is no relationship between a program and a target.

When importing a program it selects the last board you were using. It does maintain the last used for a program. @screamerbg

theotherjimmy, 2016-09-19:
Just a thought: we could include a .mbed for those examples and heed that. It would take some doing thought.

BlackstoneEngineering, 2016-09-19:
I think this would cause more confusion than it would help. A corner case that can be solved via better documentation rather than a tools issue.

jaustin, 2016-10-08:
We do actually have some places where an example is imported to the IDE with a platform hinted: 
the "Example programs" sidebar on the platform pages - for example:
https://developer.mbed.org/teams/BBC/code/microbit_blinky/?platform=Microbit

I spent a while trying to include the `platform` parameter in a <<program >> segment, but couldn't see a way to do so. I think it'd be useful in the same situations as the 'example programs' above - IE on platform pages.

sg-, 2016-10-10:
IIRC, there is an API for importing to the IDE with the platform specified. Would need to get the details from @screamerbg 

I don't recall specifically where we used it but I remember exploiting something like this a while ago

closed 2017-01-04 by andcor02