ARMmbed/mbed-os #2750: Test items fail with K64F after update

ccli8, 2016-09-20:
After update mbed-os to latest version (SHA code: eccecd3735d14230b2e3d1d57bc228992ec60c6b), many test items are failed on both K64F and NUMAKER-PFM_NUC472 boards.

## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**Toolchain version:**
4.9.3

**mbed-cli version:** 
0.9.1

**meed-os sha:**
eccecd3 Merge pull request #2733 from meriac/master

bridadan, 2016-09-20:
@ccli8 Would you be able to provide more information on what tests are failing for K64F with GCC_ARM?

ccli8, 2016-09-21:
@bridadan Attach test report for your information
```
mbedgt: test case report:
+--------------+---------------+------------------------------------------------------------------------------+--------------------------------------------------------------------------+--------+--------+---------+--------------------+
| target       | platform_name | test suite                                                                   | test case                                                                | passed | failed | result  | elapsed_time (sec) |
+--------------+---------------+------------------------------------------------------------------------------+--------------------------------------------------------------------------+--------+--------+---------+--------------------+
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-basic_test                        | Repeating Test                                                           | 2      | 0      | OK      | 0.08               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-basic_test                        | Simple Test                                                              | 1      | 0      | OK      | 0.04               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-basic_test_default                | Repeating Test                                                           | 2      | 0      | OK      | 0.07               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-basic_test_default                | Simple Test                                                              | 1      | 0      | OK      | 0.04               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Attributed Validation: Cancel Repeat                           | 1      | 0      | OK      | 0.18               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Attributed Validation: Enable Repeat Handler                   | 2      | 0      | OK      | 0.18               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Multiple Premature Validation                                  | 1      | 0      | OK      | 0.07               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Multiple Validation                                            | 1      | 0      | OK      | 0.16               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Premature Validation                                           | 1      | 0      | OK      | 0.06               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Simple Validation                                              | 1      | 0      | OK      | 0.06               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: Await                                                           | 1      | 0      | OK      | 1.41               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: CaseNext                                                        | 1      | 0      | OK      | 0.04               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: NoTimeout                                                       | 1      | 0      | OK      | 0.05               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: RepeatAllOnTimeout                                              | 1      | 0      | OK      | 0.11               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: RepeatHandlerOnTimeout                                          | 1      | 0      | OK      | 1.62               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: Timeout (Failure)                                               | 1      | 0      | OK      | 0.22               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: Timeout (Success)                                               | 1      | 0      | OK      | 0.15               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-case_control_repeat               | Control: CaseNext                                                        | 1      | 0      | OK      | 0.05               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-case_control_repeat               | Control: NoRepeat                                                        | 1      | 0      | OK      | 0.05               |
| K64F-GCC_ARM | K64F          | features-frameworks-utest-tests-unit_tests-case_control_repeat               | Control: RepeatAll                                                       | 10     | 0      | OK      | 0.05          [... cut ...]
| K64F-GCC_ARM | K64F          | tests-storage_abstraction-basicapi                                           | uninitialize                                                             | 3      | 0      | OK      | 0.09               |
+--------------+---------------+------------------------------------------------------------------------------+--------------------------------------------------------------------------+--------+--------+---------+--------------------+
mbedgt: test case results: 2 SKIPPED / 292 OK / 20 ERROR
mbedgt: completed in 5008.39 sec
mbedgt: exited with code 73
[mbed] ERROR: "mbedgt" returned error code 73.
```

sg-, 2016-09-22:
@ccli8 I dont see what change could have done that. Will have to look at closer. If you see something please let us know. @meriac @AlessandroA

bridadan, 2016-09-22:
@ccli8 Thanks for the summary, though it's still a bit tough to figure out the issue without the verbose log.

Could you rerun the tests (or just one of the failing tests) with the `-v` flag and post the log?

ccli8, 2016-09-23:
@bridadan I run the test `rtos-rtx-target_cortex_m-tests-memory-heap_and_stack` with the `-v` flag and get the following log. Besides, do you reproduce the issue with the same mbed OS SHA eccecd3735d14230b2e3d1d57bc228992ec60c6b on K64F. If your test is OK, I need to check more my environment.
```
Building project heap_and_stack (K64F, GCC_ARM)
Scan: GCC_ARM
Scan: FEATURE_IPV4
Scan: FEATURE_STORAGE
Scan: heap_and_stack
[DEBUG] Macros: -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DTARGET_Freescale -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS -DFSL_RTOS_MBED -DDEVICE_STORAGE=1 -DMBEDTLS_ENTROPY_HARDWARE_ALT -DTARGET_KPSDK_MCUS -DTOOLCHAIN_GCC -DTARGET_CORTEX_M -DTARGET_KSDK2_MCUS -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_K64F -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DCPU_MK64FN1M0VMD12 -DFEATURE_IPV4=1 -DTARGET_LIKE_MBED -DTARGET_FF_ARDUINO -DTARGET_KPSDK_CODE -DTARGET_RELEASE -DDEVICE_SERIAL_FC=1 -DFEATURE_STORAGE=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DMBED_BUILD_TIMESTAMP=1474607693.51 -DTARGET_FRDM -DDEVICE_SPI=1 -DDEVICE_ERROR_RED=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DDEVICE_PORTIN=1 -DTARGET_MCU_K64F -DARM_MATH_CM4
Link: heap_and_stack
[DEBUG] Link: C:\Program Files\GNU Tools ARM Embedded\4.9 2015q3\bin\arm-none-eabi-gcc @C:\CCLi8\IoT\mbed\mbed-os\nu-porting\mbed\.build/tests\K64F\GCC_ARM\.\rtos\rtx\TARGET_CORTEX_M\TESTS\memory\heap_and_stack\.link_files.txt
[DEBUG] Return: 0
Elf2Bin: heap_and_stack
[DEBUG] FromELF: C:\Program Files\GNU Tools ARM Embedded\4.9 2015q3\bin\arm-none-eabi-objcopy -O binary C:\CCLi8\IoT\mbed\mbed-os\nu-porting\mbed\.build/tests\K64F\GCC_ARM\.\rtos\rtx\TARGET_CORTEX_M\TESTS\memory\heap_and_stack\heap_and_stack.elf C:\CCLi8\IoT\mbed\mbed-os\nu-porting\mbed\.build/tests\K64F\GCC_ARM\.\rtos\rtx\TARGET_CORTEX_M\TESTS\memory\heap_and_stack\heap_and_stack.bin
[DEBUG] Return: 0
+-----------------+-------+-------+------+
| Module          | .text | .data | .bss |
+-----------------+-------+-------+------+
| Fill            |   117 |     0 | 2085 |
| Misc            | 34013 |  2212 |   84 |
| nu-porting/mbed | 20151 |    76 | 7367 |
| Subtotals       | 54281 |  2288 | 9536 |
+-----------------+-------+-------+------+
Allocated Heap: 65536 bytes
Allocated Stack: unknown
Total Static RAM memory (data + bss): 11824 bytes
Total RAM memory (data + bss + heap + stack): 77360 bytes
Total Flash memory (text + data + misc): 57609 bytes
Image: .build/tests/K64F/GCC_ARM/rtos/rtx/TARGET_CORTEX_M/TESTS/memory/heap_and_stack/heap_and_stack.bin

Memory map breakdown for built projects (values in Bytes):
+----------------+--------+-----------+------------+-------+------+-----------+-------------+
| name           | target | toolchain | static_ram | stack | heap | total_ram | total_flash |
+----------------+--------+-----------+------------+-------+------+-----------+-------------+
| heap_and_stack | K64F   | GCC_ARM   |          0 |     0 |    0 |         0 |        1040 |
+----------------+--------+-----------+------------+-------+------+-----------+-------------+

Build successes:
  * K64F::GCC_ARM::MBED-BUILD
  * K64F::GCC_ARM::RTOS-RTX-TARGET_CORTEX_M-TESTS-MEMORY-HEAP_AND_STACK
mbedgt: greentea test automation tool ver. 1.1.1
mbedgt: test specification file 'C:\CCLi8\IoT\mbed\mbed-os\nu-porting\mbed\.build/tests\K64F\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using 'C:\CCLi8\IoT\mbed\mbed-os\nu-porting\mbed\.build/tests\K64F\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
	+---------------+----------------------+-------------+-------------+--------------------------+
	| platform_name | platform_name_unique | serial_port | mount_point | target_id                |
	+---------------+----------------------+-------------+-------------+--------------------------+
	| K64F          | K64F[0]              | COM12       | D:          | 02400201C31F3E103EE1C3A8 |
	+---------------+----------------------+-------------+-------------+--------------------------+
mbedgt: processing target 'K64F' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
	+---------------+----------------------+-------------+-------------+--------------------------+
	| platform_name | platform_name_unique | serial_port | mount_point | target_id                |
	+---------------+----------------------+-------------+-------------+--------------------------+
	| K64F          | K64F[0]              | COM12:9600  | D:          | 02400201C31F3E103EE1C3A8 |
	+---------------+----------------------+-------------+-------------+--------------------------+
mbedgt: test case filter (specified with -n option)
	test filtered in 'rtos-rtx-target_cortex_m-tests-memory-heap_and_stack'
mbedgt: running 1 test for platform 'K64F' and toolchain 'GCC_ARM'
	use 1 instance for testing
mbedgt: checking for 'host_tests' directory above image directory structure
	'host_tests' directory not found: two directory levels above image path checked
mbedgt: selecting test case observer...
	calling mbedhtrun: mbedhtrun -m K64F -p COM12:9600 -f ".build/tests/K64F/GCC_ARM/rtos/rtx/TARGET_CORTEX_M/TESTS/memory/heap_and_stack/heap_and_stack.bin" -d D: -C 4 -c shell -t 02400201C31F3E103EE1C3 [... cut ...]
[mbed] Working path "C:\CCLi8\IoT\mbed\mbed-os\nu-porting\mbed" (program)
[mbed] Exec "python -u C:\CCLi8\IoT\mbed\mbed-os\nu-porting\mbed\tools\test.py -t GCC_ARM -m K64F --source C:\CCLi8\IoT\mbed\mbed-os\nu-porting\mbed --build C:\CCLi8\IoT\mbed\mbed-os\nu-porting\mbed\.build/tests\K64F\GCC_ARM --test-spec C:\CCLi8\IoT\mbed\mbed-os\nu-porting\mbed\.build/tests\K64F\GCC_ARM\test_spec.json -n rtos-rtx-target_cortex_m-tests-memory-heap_and_stack -v" in C:\CCLi8\IoT\mbed\mbed-os\nu-porting\mbed
```

ccli8, 2016-09-26:
Failed test items are fixed after the python module **mbed-greentea** is upgraded from 1.1.1 to 1.2.1.

closed 2016-09-26 by ccli8

bridadan, 2016-09-26:
@ccli8 Thanks for reporting the fix!