ARMmbed/mbed-os #2755: Export of mbed-dev to CooCox IDE cannot compile, `-std=gnu++98` is not valid for C

janjongboom, 2016-09-20:
From a user:

> I tried to compile mbed official/mbed-dev (https://developer.mbed.org/users/mbed_official/code/mbed-dev/
) code offline in coocox coide ide. I exported this code as a program to the offline ide. But when I build the code, I got 2 same errors 100s of time. Errors are:
>
>
>[cc] cc1.exe: warning: command line option '-std=gnu++98' is valid for C++/ObjC++ but not for C
>
>[cc] cc1.exe: warning: command line option '-fno-rtti' is valid for C++/ObjC++ but not for C
>
>[cc] cc1.exe: warning: command line option '-std=gnu++98' is valid for C++/ObjC++ but not for C
>
>[cc] cc1.exe: warning: command line option '-fno-rtti' is valid for C++/ObjC++ but not for C
>
>I found in configuration that if I remove -fno-rtti and -std=gnu++98 and replace it with -std=gnu90 those errors get rem oved. But as these settings are done by ARM their should be some reason and if i delete these i might face some problems in future.

I think something goes wrong in the exporter, should be targeting C++ not C...

@theotherjimmy

janjongboom, 2016-09-20:
Hmm... let's first ask user if what he's doing actually makes sense...

janjongboom, 2016-09-20:
Verified. Real issue.

theotherjimmy, 2016-09-20:
The coide project files are all target specific. To fix this issue, you would have to change every line like this one: https://github.com/ARMmbed/mbed-os/blob/master/tools/export/coide_lpc1768.coproj.tmpl#L9 to account for the difference between C and C++ files. I'm not entirely sure what the coide project file structure looks like, so I can't comment on if this is possible in that IDE. If only we could get it to just run make behind the scenes like Eclipse...

sg-, 2016-09-22:
coide templates needs a refactor. Will consider this an enhancement.

closed 2017-10-27 by ChiefBureaucraticOfficer