ARMmbed/mbed-os #2758: Mbed 2 - Mbed OS 5 : Sleep incorrect for Nordic target

aracosta1, 2016-09-20:
### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
TARGET_NORDIC
TARGET_MCU_NRF51822
TARGET_NRF5

The sleep() function don't take into account the use of the softdevice.
the correct implementation should be similar to the one used on [mbed OS 3:](https://github.com/ARMmbed/mbed-hal-nrf51822-mcu/blob/master/source/sleep.c#L26)

pan-, 2016-09-21:
@nvlsianpu Could you look at this ?

nvlsianpu, 2016-09-21:
Thanks for yours comment. I will take care of fix it.

adbridge, 2016-10-26:
@aracosta1 It looks like your issue should be fixed with the latest release of mbed-os (5.2.0). Could you please verify and close this issue if so ? Thanks.

aracosta1, 2016-11-08:
I'm doing some basic test and it seem to work correctly.
I had one warning during a compilation obout sleep.o not unique but now I cannot reproduce te issue anymore.
I will try to do some test also on OS version 2

aracosta1, 2016-11-18:
At the moment I cannot test for the OS version 2 because the build  system is broken for the latest library version  (see bug  #3249). I will update when the build system is fixed

danielknox, 2016-11-25:
Hi, I've noticed since 5.2.0 the power consumption of the NRF51 has increased and I believe this is related to the change in the sleep code. 

I've been referring to the Nordic guide for measuring power consumption on the board to get an approximate idea of my board's power requirements. This consists of a 10ohm resistor and using a oscilloscope to measure the differential voltage on the current measurement pins of the dev kit. Roughly this means 10mV to 1mA. 

This simple test program (http://pastebin.com/chP5S5Uk) reads 1.2mV for 5.1 of Mbed OS, but increases to 10mV in subsequent versions. Whilst this seems small, as I'm trying to run this on a production board off a coin cell, this is a huge difference!

Any advice would be greatly appreciated.

aracosta1, 2016-12-21:
I had done some test and for me it's working so I close the issue.

closed 2016-12-21 by aracosta1