ARMmbed/mbed-os #2769: Call __WFI() when going idle

betzw, 2016-09-21:
### Description
- Type: Enhancement
- Priority: Minor

---------------------------------------------------------------
## Enhancement
As long as there is no tickless mode, function `default_idle_hook()` should (at least) call `__WFI()` to reduce power consumption.

**Target**
all

**Toolchain:**
all

**mbed-cli version:** 
all

**meed-os sha:**Incidental Remark: Do **not** call `sleep()` as it would disable suspend the HAL tick (at least on some platforms).
bdab10d Merge pull request #2590 from ARMmbed/release

**Expected behavior**
Reduced power consumption when idle.

**Actual behavior**
No power consumption when idle.

----------------------------------------------------------------
## Enhancement

**Suggested enhancement**
Call `__WFI()` inside function `default_idle_hook()` in non-tickless mode.

**Incidental Remark**
Do **not** call `sleep()` as it would suspend the HAL tick (at least on some platforms).

sg-, 2016-09-22:
@betzw can you point at the particular change? bdab10d is quite large

betzw, 2016-09-23:
It is not related to any particular change since it seems as if this behavior exists since a long time.
Basically it is about commenting the call to sleep out in funcrion ` default_idle_hook()` (see [here](https://github.com/ARMmbed/mbed-os/blob/master/rtos/rtos/rtos_idle.c#L31)).

0xc0170, 2016-09-23:
Tickless mode shall get into the code base this year, there were some problems as you might have noticed with tickers that needs fixing, it's parked on the feature branch at the moment.

Even if you add ``__WFI()`` how much would we save as systick is running?

betzw, 2016-09-23:
No, I didn't notice the issues with tickers. But did I get it right that when calling `__WFI()` in some situations tickers might not behave correctly anymore? If yes, obviously you shouldn't call `__WFI()` for the moment, otherwise even if it will save less than in tickless mode, it will anyway improve power consumption.

betzw, 2016-11-14:
Any news on this @sg-?

betzw, 2016-12-19:
Ping!

0xc0170, 2016-12-19:
There's feature branch, that needs more work: https://github.com/ARMmbed/mbed-os/tree/feature_rtos_tickless. It should resolve this.

betzw, 2016-12-20:
Seems like an abandoned branch, I mean last commit about 4 months ago. Have you any timeline for when this branch would be merged?

betzw, 2016-12-20:
@screamerbg

bulislaw, 2016-12-20:
There were difficulties with the tickless mode due to differences in platforms and how they implement low power timer. I have an idea how to address it, I'll write it down over Christmas and then we can discuss it and find some time to implement it.

0xc0170, 2016-12-20:
> Seems like an abandoned branch, I mean last commit about 4 months ago. Have you any timeline for when this branch would be merged?

Addition to the comment above about issues with the tickers, there's a plan for RTX update, therefore this idle loop work should be based on that update (version >5.4)

bulislaw, 2016-12-20:
That shouldn't be a problem, as the RTX API changes are cosmetic.

closed 2017-03-22 by betzw