ARMmbed/mbed-os #2783: NUCLEO_F091RC: Multiple test failures

bulislaw, 2016-09-22:
### Description
- Type: Bug

## Bug
**Target**
NUCLEO_F091RC

**Toolchain:**
GCC_ARM|ARM

**Toolchain version:**
GCC: 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
ARM: ARMCompiler5.06u2
**mbed-cli version:** 
0.9.1

**meed-os sha:**
5d1d8b4

Multiple failures in automated tests both for MBED2 and MBED5:

**MBED5: ARM_GCC**
```
mbedgt: test case report:
+-----------------------+---------------+------------------------------------------------------------------------------+---------------------------------------------------------------+--------+--------+---------+--------------------+
| target                | platform_name | test suite                                                                   | test case                                                     | passed | failed | result  | elapsed_time (sec) |
+-----------------------+---------------+------------------------------------------------------------------------------+---------------------------------------------------------------+--------+--------+---------+--------------------+
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-basic_test                        | Repeating Test                                                | 2      | 0      | OK      | 0.07               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-basic_test                        | Simple Test                                                   | 1      | 0      | OK      | 0.04               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-basic_test_default                | features-frameworks-utest-tests-unit_tests-basic_test_default | 0      | 1      | ERROR   | 70.0               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Attributed Validation: Cancel Repeat                | 1      | 0      | OK      | 0.18               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Attributed Validation: Enable Repeat Handler        | 2      | 0      | OK      | 0.18               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Multiple Premature Validation                       | 1      | 0      | OK      | 0.07               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Multiple Validation                                 | 1      | 0      | OK      | 0.16               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Premature Validation                                | 1      | 0      | OK      | 0.05               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_async_validate               | Validate: Simple Validation                                   | 1      | 0      | OK      | 0.06               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: Await                                                | 1      | 0      | OK      | 1.41               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: CaseNext                                             | 1      | 0      | OK      | 0.04               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: NoTimeout                                            | 1      | 0      | OK      | 0.04               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: RepeatAllOnTimeout                                   | 1      | 0      | OK      | 0.11               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: RepeatHandlerOnTimeout                               | 1      | 0      | OK      | 1.62               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: Timeout (Failure)                                    | 1      | 0      | OK      | 0.21               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_control_async                | Control: Timeout (Success)                                    | 1      | 0      | OK      | 0.15               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_control_repeat               | Control: CaseNext                                             | 1      | 0      | OK      | 0.04               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_control_repeat               | Control: NoRepeat                                             | 1      | 0      | OK      | 0.04               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_control_repeat               | Control: RepeatAll                                            | 10     | 0      | OK      | 0.05               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_control_repeat               | Control: RepeatHandler                                        | 10     | 0      | OK      | 0.06               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_selection                    | features-frameworks-utest-tests-unit_tests-case_selection     | 0      | 1      | ERROR   | 69.78              |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_setup_failure                | Setup handler returns ABORT                                   | 1      | 0      | OK      | 0.13               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_setup_failure                | Setup handler returns CONTINUE                                | 1      | 0      | OK      | 0.06               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_setup_failure                | Setup handler returns IGNORE                                  | 1      | 0      | OK      | 0.13               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_teardown_failure             | Teardown handler returns ABORT                                | 1      | 0      | OK      | 0.06               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_teardown_failure             | Teardown handler returns ABORT but is IGNORED                 | 1      | 0      | OK      | 0.08               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-case_teardown_failure             | Teardown handler returns CONTINUE                             | 1      | 0      | OK      | 0.05               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-control_type                      | Testing combinations of different group                       | 1      | 0      | OK      | 0.06               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-control_type                      | Testing combinations of same group                            | 1      | 0      | OK      | 0.07               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-control_type                      | Testing constants                                             | 1      | 0      | OK      | 0.05               |
| NUCLEO_F091RC-GCC_ARM | NUCLEO_F091RC | features-frameworks-utest-tests-unit_tests-control_type                      | Testing constructors                                          | 1      | 0      | OK      | 0.05               |
[... cut ...]
| OK     | NUCLEO_F091RC[9340] | ARM       | RTOS_6      | Mail messaging                        |        2.2         |       20      |  1/1  |
| OK     | NUCLEO_F091RC[9340] | ARM       | RTOS_7      | Timer                                 |       11.48        |       15      |  1/1  |
| OK     | NUCLEO_F091RC[9340] | ARM       | RTOS_8      | ISR (Queue)                           |        6.57        |       20      |  1/1  |
+--------+---------------------+-----------+-------------+---------------------------------------+--------------------+---------------+-------+
Result: 1 FAIL / 26 OK

```

jeromecoutant, 2016-09-23:
Hi

I have checked on master branch with NUCLEO_F091RC,
All tests are now OK except:
- MBED OS2 MBED_37/MBED_38 are still FAIL
- MBED OS5 lp_timeout and lp_ticker tests, please wait for: https://github.com/ARMmbed/mbed-os/pull/2759

Regards,

jeromecoutant, 2016-11-30:
Maybe we can close this issue ?
Thx

0xc0170, 2016-11-30:
Is MBED OS2 MBED_37/MBED_38 are still FAIL tracked anywhere? (is it still relevant?)

jeromecoutant, 2016-11-30:
Results from ST test bench:
- with mbed_lib_rev128: tests are OK for ARM and uARM, FAIL for GCC and IAR
- with mbed_lib_rev129: tests are OK for all (ARM, uARM, GCC and IAR)

adustm, 2017-01-03:
Hello, 
. with mbed_lib_rev132, MBED_37/MBED_38 tests are OK for IAR / GCC_ARM / ARM
Can we close this issue?
Cheers

bridadan, 2017-01-03:
@bulislaw If you're happy could you please close the issue? Thanks.

bulislaw, 2017-01-03:
Closing, thanks!

closed 2017-01-03 by bulislaw