ARMmbed/mbed-os #2784: Build system does not allow nested features.

SeppoTakalo, 2016-09-22:
### Description
- Type: Bug
- Priority: Blocker.
I cannot create Nanostack binary release because of this.

---------------------------------------------------------------
## Bug

**Expected behavior**
I should be able to create nested features like `FEATURE_NANOSTACK/FEATURE_LOWPAN_ROUTER`

**Actual behavior**
Build system is not picking up the second level of features.

**Steps to reproduce**
Create a file a `FEATURE_X` directory under any known feature. X should be on of allowed from `config.py`
Files under this `FEATURE_X` are ignored by build system.

SeppoTakalo, 2016-09-22:
CC: @0xc0170 @sg- @kjbracey-arm

SeppoTakalo, 2016-09-22:
CC: @TuomoHautamaki

theotherjimmy, 2016-09-22:
Yes: This is simply that the Resources object has a Resources object for each feature and assumes that there will be no features nested within those feature objects. Fixing this would require that the build system traverse a DAG of resources objects to collect it's build files. @screamerbg

sg-, 2016-09-22:
Can we just remove the top level ```FEATURE_NANOSTACK``` and rename to ```nanostack```?

SeppoTakalo, 2016-09-23:
Removing the top level `FEATURE_NANOSTACK` works fine.

I'll continue with that and send PR soon.

SeppoTakalo, 2016-09-26:
CC: @sam-taylor-arm @sg- @rhenson @TuomoHautamaki @tommihei01 

This should now be considered as a blocker bug for us.
We cannot do the binary release as we have planned to do it.

```
+
|
+-+FEATURE_NANOSTACK
|         +
|         +---+FEATURE_LOWPAN_ROUTER
|              FEATURE_LOWPAN_HOST
|              ...
|              FEATURE_THREAD_ROUTER
|
+--+FEATURE_COMMON_PAL
```

Build system needs to be fixed to handle features a same way as `TOOLCHAIN_*` and `TARGET_*` are handled. They allow nesting.

theotherjimmy, 2016-12-20:
So, it seems that the nanostack binary has been updated despite this. I'm going to close this issue. Please re-open if you have another use-case that would be broken by this behavior.

SeppoTakalo, 2016-12-21:
I don't agree the reason for closing this.
Features don't work as expected, therefore this bug.

I would expect features work in a way how `TARGET_*` and `TOOLCHAIN_*` would work. I would like to use the behaviour similar to `TARGET_*` so that I can, for example, drop RF drivers from example applications when Nanostack is not compiled in.

If we were able to make a workaround and release Nanostack regardless of this bug, it is not a reason for closing this.

Either fix the issue, or document how the FEATURE_* thing works.

closed 2017-01-18 by sg-