ARMmbed/mbed-os #2786: Question: have there been any recent CMSIS RTOS initialisation/memory/behaviour changes?

RobMeades, 2016-09-22:
I'm in the process of integrating a new target into mbed-os.  A week ago, at #7669d7f, I had a working Pull Request that passed all the `htrun` tests ready to go.  Because of RTX file change collisions I've had to rebase to #c1c1492 and now CMSIS RTOS doesn't even get into `mbed_sdk_init()`: `rt_tsk_create()` returns 0.

Can anyone point me at any recent changes to do with memory configuration, initialisation or RTOS behaviour before I start diffing the world and tearing my hair out?

Rob

0xc0170, 2016-09-23:
Looking at the history, https://github.com/ARMmbed/mbed-os/commits/master/rtos , there's stack check enabled, some targets fixes, Thread class clean up.

RobMeades, 2016-09-23:
Thanks Martin.  It's definitely one of those as, if I revert the changes in that directory, my target works again.  Digging...

RobMeades, 2016-09-23:
It appears to be the [Add RTX API to report stack usage](https://github.com/ARMmbed/mbed-os/commit/c31929669265b51abd2d3a5102a161fdcf77108d) commit that's causing my problem.  Investigating...

RobMeades, 2016-09-23:
If I include the `void *argv` parameter in the `OS_TCB` struct in `rt_TypeDef.h`, `rt_task_create()` fails (even if I comment out all uses of `argv`) , if I add it back in then `rt_task_create()` returns 1 and things work.  So something doesn't fit with these extra 4 bytes in the TCB.

RobMeades, 2016-09-23:
It looks as though there might be an issue with #2642 (at least, there is for me).  With the `void *argv` pointer added to the `OS_TCB` struct it is now 64 bytes in size.  In `RTX_CM_lib.h` I see  '#define OS_TCB_SIZE 60', which I'm guessing defines the size of this struct (why not use `sizeof()`?).  From the map file, with this pointer added, the struct is now 64 bytes big.  With the #define left as it is `rt_task_create()` returns 0, if I increase the #define to 64 `rt_task_create()` returns 1 and the RTOS starts.

Now I'm not exactly sure what I'm doing here as this code is somewhat obtuse, and I've no idea why this isn't a problem for anyone else, so I'm very happy to be told that it's actually my problem but I do think I need someone else to take a look now.  Help!

RobMeades, 2016-09-24:
I've proposed changing `OS_TCB_SIZE` to 64 in the above pull request.  If a different fix is required, please feel free to reject the pull request.

0xc0170, 2016-09-26:
Thanks for resolving this.

closed 2016-09-26 by 0xc0170