ARMmbed/mbed-os #2789: post build scripts not exportable

maclobdell, 2016-09-22:
### Description
- Type: Enhancement
- Related issue: n/a
- Priority:  Major

## Enhancement

**Reason to enhance or problem with existing solution**

Targets such as NCS36510 require post-build scripts to perform target-specific actions such as calculate CRC, add firmware revision number, and place in a specific location in memory.  At this time, this is accomplished in mbed cli by kicking off a python script from target.py.  

However, this sub-script is not standalone and cannot be exported to work with Eclipse, IAR, Keil, etc.  This presents a headache when developing in those IDEs if they don't have an alternative post build for the target.  

**Suggested enhancement**

It would be nice if post build scripts were stand-alone and just act on the elf, bin, or hex files that came out of the linker.  Then they could be called directly by a makefile or IDE settings.  

**Pros**
IDEs would  support building for target with post build scripts easily.

**Cons**
Would change architecture of build system (slightly?)

@theotherjimmy

0xc0170, 2016-09-23:
> However, this sub-script is not standalone and cannot be exported to work with Eclipse, IAR, Keil, etc. This presents a headache when developing in those IDEs if they don't have an alternative post build for the target. 

I am curious how they ship the example projects for various IDE, what do they provide (what type of postbuild script do they add to the projects) ?

maclobdell, 2016-09-29:
You make a good point.  MCU vendors should work with the IDE vendors to get support for post-build, flashing, etc.  We can't do it all.

closed 2016-09-29 by maclobdell