ARMmbed/mbed-os #2796: NUCLEO_F334R8: Multiple test failures

bulislaw, 2016-09-23:
### Description
- Type: Bug

## Bug
**Target**
NUCLEO_F334R8
**Toolchain:**
GCC_ARM|ARM

**Toolchain version:**
GCC: 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
ARM: ARMCompiler5.06u2
**mbed-cli version:** 
0.9.1

**meed-os sha:**
c1c1492

Multiple failures in automated tests both for MBED2 and MBED5:
**MBED2 GCC**
```
+---------------+---------------------+-----------+-------------+---------------------------------------+--------------------+---------------+-------+
| Result        | Target              | Toolchain | Test ID     | Test Description                      | Elapsed Time (sec) | Timeout (sec) | Loops |
+---------------+---------------------+-----------+-------------+---------------------------------------+--------------------+---------------+-------+
| OK            | NUCLEO_F334R8[9422] | GCC_ARM   | DTCT_1      | Simple detect test                    |        0.12        |       10      |  1/1  |
| OK            | NUCLEO_F334R8[9422] | GCC_ARM   | EXAMPLE_1   | /dev/null                             |        0.12        |       10      |  1/1  |
| OK            | NUCLEO_F334R8[9422] | GCC_ARM   | MBED_10     | Hello World                           |        0.12        |       10      |  1/2  |
| OK            | NUCLEO_F334R8[9422] | GCC_ARM   | MBED_11     | Ticker Int                            |        0.12        |       20      |  1/1  |
| OK            | NUCLEO_F334R8[9422] | GCC_ARM   | MBED_12     | C++                                   |        1.27        |       10      |  1/1  |
| OK            | NUCLEO_F334R8[9422] | GCC_ARM   | MBED_16     | RTC                                   |        4.49        |       20      |  1/1  |
| OK            | NUCLEO_F334R8[9422] | GCC_ARM   | MBED_2      | stdio                                 |        0.9         |       20      |  1/2  |
| OK            | NUCLEO_F334R8[9422] | GCC_ARM   | MBED_23     | Ticker Int us                         |       11.22        |       15      |  1/1  |
| OK            | NUCLEO_F334R8[9422] | GCC_ARM   | MBED_24     | Timeout Int us                        |       11.82        |       15      |  1/1  |
| OK            | NUCLEO_F334R8[9422] | GCC_ARM   | MBED_25     | Time us                               |       11.44        |       15      |  1/2  |
| OK            | NUCLEO_F334R8[9422] | GCC_ARM   | MBED_26     | Integer constant division             |        1.49        |       20      |  1/1  |
| OK            | NUCLEO_F334R8[9422] | GCC_ARM   | MBED_34     | Ticker Two callbacks                  |       11.54        |       15      |  1/3  |
| FAIL          | NUCLEO_F334R8[9422] | GCC_ARM   | MBED_37     | Serial NC RX                          |        5.85        |       20      |  0/3  |
| FAIL          | NUCLEO_F334R8[9422] | GCC_ARM   | MBED_38     | Serial NC TX                          |        5.67        |       20      |  0/3  |
| OK            | NUCLEO_F334R8[9422] | GCC_ARM   | MBED_A1     | Basic                                 |        1.12        |       10      |  1/1  |
| OK            | NUCLEO_F334R8[9422] | GCC_ARM   | MBED_A21    | Call function before main (mbed_main) |        1.26        |       20      |  1/2  |
| OK            | NUCLEO_F334R8[9422] | GCC_ARM   | MBED_A28    | CAN loopback test                     |       10.43        |       20      |  1/1  |
| OK            | NUCLEO_F334R8[9422] | GCC_ARM   | MBED_A9     | Serial Echo at 115200                 |        6.63        |       20      |  1/2  |
| OK            | NUCLEO_F334R8[9422] | GCC_ARM   | MBED_BUSOUT | BusOut                                |        2.12        |       15      |  1/1  |
| NOT_SUPPORTED | NUCLEO_F334R8       | GCC_ARM   | RTOS_1      | Basic thread                          |         0          |       0       |   -   |
| NOT_SUPPORTED | NUCLEO_F334R8       | GCC_ARM   | RTOS_2      | Mutex resource lock                   |         0          |       0       |   -   |
| NOT_SUPPORTED | NUCLEO_F334R8       | GCC_ARM   | RTOS_3      | Semaphore resource lock               |         0          |       0       |   -   |
| NOT_SUPPORTED | NUCLEO_F334R8       | GCC_ARM   | RTOS_4      | Signals messaging                     |         0          |       0       |   -   |
| NOT_SUPPORTED | NUCLEO_F334R8       | GCC_ARM   | RTOS_5      | Queue messaging                       |         0          |       0       |   -   |
| NOT_SUPPORTED | NUCLEO_F334R8       | GCC_ARM   | RTOS_6      | Mail messaging                        |         0          |       0       |   -   |
| NOT_SUPPORTED | NUCLEO_F334R8       | GCC_ARM   | RTOS_7      | Timer                                 |         0          |       0       |   -   |
| NOT_SUPPORTED | NUCLEO_F334R8       | GCC_ARM   | RTOS_8      | ISR (Queue)                           |         0          |       0       |   -   |
+---------------+---------------------+-----------+-------------+---------------------------------------+--------------------+---------------+-------+
Result: 2 FAIL / 17 OK / 8 NOT_SUPPORTED

```
**MBED2 ARM**
```
Test summary:
+---------+---------------------+-----------+-------------+---------------------------------------+--------------------+---------------+-------+
| Result  | Target              | Toolchain | Test ID     | Test Description                      | Elapsed Time (sec) | Timeout (sec) | Loops |
+---------+---------------------+-----------+-------------+---------------------------------------+--------------------+---------------+-------+
| OK      | NUCLEO_F334R8[9422] | ARM       | DTCT_1      | Simple detect test                    |        0.39        |       10      |  1/2  |
| FAIL    | NUCLEO_F334R8[9422] | ARM       | EXAMPLE_1   | /dev/null                             |        0.15        |       10      |  0/3  |
| OK      | NUCLEO_F334R8[9422] | ARM       | MBED_10     | Hello World                           |        0.12        |       10      |  1/2  |
| OK      | NUCLEO_F334R8[9422] | ARM       | MBED_11     | Ticker Int                            |        0.16        |       20      |  1/1  |
| OK      | NUCLEO_F334R8[9422] | ARM       | MBED_12     | C++                                   |        1.52        |       10      |  1/2  |
| OK      | NUCLEO_F334R8[9422] | ARM       | MBED_16     | RTC                                   |        4.73        |       20      |  1/3  |
| OK      | NUCLEO_F334R8[9422] | ARM       | MBED_2      | stdio                                 |        0.82        |       20      |  1/2  |
| FAIL    | NUCLEO_F334R8[9422] | ARM       | MBED_23     | Ticker Int us                         |        0.15        |       15      |  0/3  |
| FAIL    | NUCLEO_F334R8[9422] | ARM       | MBED_24     | Timeout Int us                        |        30.1        |       15      |  0/3  |
[... 133 lines cut ...]
| NUCLEO_F334R8-GCC_ARM | NUCLEO_F334R8 | rtos-rtx-target_cortex_m-tests-memory-heap_and_stack                         | rtos-rtx-target_cortex_m-tests-memory-heap_and_stack                         | 0      | 1      | ERROR  | 65.64              |
| NUCLEO_F334R8-GCC_ARM | NUCLEO_F334R8 | tests-integration-basic                                                      | tests-integration-basic                                                      | 0      | 1      | ERROR  | 65.53              |
| NUCLEO_F334R8-GCC_ARM | NUCLEO_F334R8 | tests-mbed_drivers-c_strings                                                 | tests-mbed_drivers-c_strings                                                 | 0      | 1      | ERROR  | 67.05              |
| NUCLEO_F334R8-GCC_ARM | NUCLEO_F334R8 | tests-mbed_drivers-dev_null                                                  | tests-mbed_drivers-dev_null                                                  | 0      | 1      | ERROR  | 66.0               |
| NUCLEO_F334R8-GCC_ARM | NUCLEO_F334R8 | tests-mbed_drivers-echo                                                      | tests-mbed_drivers-echo                                                      | 0      | 1      | ERROR  | 65.96              |
| NUCLEO_F334R8-GCC_ARM | NUCLEO_F334R8 | tests-mbed_drivers-generic_tests                                             | tests-mbed_drivers-generic_tests                                             | 0      | 1      | ERROR  | 66.92              |
[... cut ...]
| NUCLEO_F334R8-GCC_ARM | NUCLEO_F334R8 | tests-mbedmicro-mbed-div                                                     | tests-mbedmicro-mbed-div                                                     | 0      | 1      | ERROR  | 66.14              |
+-----------------------+---------------+------------------------------------------------------------------------------+------------------------------------------------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 36 ERROR
```

jeromecoutant, 2016-09-23:
Hi
NUCLEO_F334R8 is not MBED OS5 compatible.
I am going to check MBED OS2 failed tests

bulislaw, 2016-09-23:
Ah right it's listed as MBED 2 only. Thanks.

jeromecoutant, 2016-11-30:
Maybe we can close this issue ?
Thx

jeromecoutant, 2016-11-30:
Results from ST test bench:
- with mbed_lib_rev129: all tests are OK for all tool chains (ARM, uARM, GCC and IAR)

adustm, 2017-01-03:
Hello,
Jerome's results are confirmed on the mbed_lib_rev132.

Can someone close this issue ?
Cheers
Armelle

closed 2017-01-03 by bulislaw